rp-rs/rp-hal #10: SPIsupport

neilgall, 2021-01-30:
There are two SPI interfaces described in section 4.4 of the [datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf)

Both interfaces support:
* Master or slave mode
* Programmable clock and word size
* Programmable choice of GPIO pins
* Separate transmit, receive, overflow and receive timeout interrupts

hmvp, 2021-06-29:
I'd like to make start with this if nobody minds..

neilgall, 2021-06-29:
Yes, I started back in February but haven't had much time.

anall, 2021-07-02:
@hmvp I was actually thinking of looking at this as an example of something that makes use of the fancy GPIO refactor, but if you've made progress or really want to work on this feel free to continue and I'll refactor afterward — but I can't guarantee how much will be left.

hmvp, 2021-07-02:
@anall I pushed what I have so far in https://github.com/rp-rs/rp-hal/pull/50 I don't own a scope so its a bit hard to figure out why it is not working properly. So I don't mind you finishing it.

I was thinking on rebasing it to your gpio branch since i needed funcsel.

anall, 2021-07-02:
That's actually a decent start. I left a review on #50 but minimal changes were needed to make it work.

9names, 2021-07-24:
Initial implementation merged in #50

closed 2021-07-24 by 9names