rp-rs/rp-hal #4: Add control for GPIOs

tdittr, 2021-01-30:
The datasheet covers GPIOs in chapter 2.19.

# Use cases
The GPIOs can be used for these functions (chapter 2.19.2):
- Control/read pin level (SIO)
- Connecting a pin to a peripheral (PIO, SPI, UART, I2C, PWM, ADC, USB Vbus control)
- Use as an interrupt (chapter 2.19.3)

The following settings can be changed for each pin (chapter 2.19.4):
- Output drive strength
- Output slew rate
- Schmitt trigger
- Pull-up/down
- Signal and direction inversion and override

The input buffer and output driver can be disabled (which is mandatory when used with the ADC (chapter 4.9.1)).

# Implementation ideas
## Type states
To make it harder to misuse we could add the following as type parameters (similar to [NRF HAL](https://github.com/nrf-rs/nrf-hal/blob/master/nrf-hal-common/src/gpio.rs)):
- Function selection (In-/Output (SIO), PIO[01], SPI[01]-(Rx|Tx|...), ...)
- Pull-Up/Down (only for Inputs)

## Traits
All pins have the same configurations listed above. So a trait that covers all these settings could make sense.

## Interrupts
ToDo...

anall, 2021-05-08:
I know the basic support just landed a few days ago, but I think refactoring this to be more like how the `atsamd` HAL does things (which seems to me to be one of the most mature HAL crates)

I hope I am not stepping on anyone's feet, but I've got a "minimum" viable example of my refactoring that I'll open as a draft PR.

9names, 2021-07-24:
GPIO and pin muxing complete as #34 merged

closed 2021-07-24 by 9names