rp-rs/rp-hal #6: Allow setup of the clocks block

tdittr, 2021-01-30:
The clock systems are described in chapter 1.15 of the [datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf).

# Use cases
The clock block provides clocks to all parts of the SoC. Thus it defines how much time equals one clock cycle for different parts of the chip.

## Clock sources
- Ring Oscilator (ROSC, chapter 2.17)
- Crystal Oscilator (XOSC, chapter 2.16)
- USB- and System-PLL (referencing XOSC, chapter 2.18)
- External GPIO 

## Switching clock sources
Clock glitches need to be avoided when changing the clock source, see chapter 2.15.3.2.

Also changing the clock speed will affect all connected components. E. g. halfing the divider for `clk_peri` would double the baud rate of a running UART.

# Implementation ideas
- Allow `frreze`ing the clock speed. This should consume the clock path so it can not be changed while a device is using it. Example: [stm32f30x-hal](https://github.com/japaric/stm32f30x-hal/blob/master/src/rcc.rs#L144) But maybe there should be a way to get back control over the clock speeds.

tdittr, 2021-01-31:
I will start implementing a proof-of-concept for one clock path. (XOSC -> SYS PLL -> clk_sys -> Frequency Counter)

Although I will have to wait for my Pico to arrive before I can test it.

tdittr, 2021-02-07:
I started working on the XOsc in #18

tdittr, 2021-03-15:
Hi, I don't have time to finish this. If anybody wants to pick up from here I would be happy to help and review.

9names, 2021-06-27:
This was picked up in https://github.com/rp-rs/rp-hal/pull/31 and got merged in https://github.com/rp-rs/rp-hal/pull/46

closed 2021-06-27 by 9names