rp-rs/rp-hal #8: I2C Support

eolder, 2021-01-30:
See Chapter 4 Section 3 of the [datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf) for more information.

Each  I2C  controller  is  based  on  a  configuration  of  the  Synopsys  DW_apb_i2c  (v2.01)  IP.  The  following  features  are supported:
•Master or Slave (Default to Master mode)
•Standard mode, Fast mode or Fast mode plus
•Default slave address 0x055
•Supports 10-bit addressing in Master mode
•16-element transmit buffer
•16-element receive buffer
•Can be driven from DMA
•Can generate interrupts

##Implementation

TODO

leo60228, 2021-07-04:
@AsafFisher Are you still working on this?

AsafFisher, 2021-07-04:
no

9names, 2021-07-24:
Initial implementation merged in #56

closed 2021-07-24 by 9names