rp-rs/rp-hal #13: DMA on the way!

AsafFisher, 2021-02-02:
Implement the DMA api for everyone :memo:

tdittr, 2021-02-03:
Things I had on my list for the DMA:
- Example: https://docs.rs/stm32f1xx-hal/0.7.0/stm32f1xx_hal/dma/index.html
- Traits: https://crates.io/crates/embedded-dma

AsafFisher, 2021-02-03:
Thank you for the guidance! I will look in to it (:

AsafFisher, 2021-03-11:
I dont like `https://crates.io/crates/embedded-dma`'s traits why should I follow them?

tdittr, 2021-03-11:
Well, I guess it would allow more interoperation, similar to the embedded-hal traits.

maximeborges, 2021-11-10:
What is the status of this implementation?

9names, 2021-11-13:
@maximeborges The branch where this work took place is https://github.com/rp-rs/rp-hal/tree/rp-dma
It has not been touched in 7 months. I would call the status of this implementation "abandoned".
Use it as a starting point if you wish, but if you a need for DMA I suggest starting a new branch and working on it rather than waiting for this to be finished.

mgottschlag, 2021-12-15:
See #209 for a more complete take on DMA which is already usable.

lightsing, 2022-08-21:
How is it going? Is there any working on progress?

9names, 2022-08-22:
No progress

eloraiby, 2022-09-22:
@AsafFisher any news on this one?

9names, 2022-09-23:
AsafFisher has not been active on this project in over a year.
No progress to report

ogoffart, 2022-09-23:
I have been using the PR #209 in my project (details in this comment: https://github.com/rp-rs/rp-hal/pull/209#issuecomment-1156582772 ) I have rebased the change to the recent branch in https://github.com/slint-ui/rp-hal/tree/dma-0.6

9names, 2022-09-23:
I did a rebase of your rebase earlier today.
https://github.com/9names/rp-hal/tree/dma-0.6
The plan was going to write + run some tests to see what works, maybe clean up the implementation and write some docs

ogoffart, 2022-09-23:
That would be nice to see that merged. I also tried to upgrade to embedded_dma 0.2 in https://github.com/slint-ui/rp-hal/tree/embedded_dma2 (which i just pushed now),  but i haven't tried it yet.

9names, 2023-06-24:
Closed via #457

closed 2023-06-24 by 9names