rp-rs/rp-hal #14: Blinky example?

btrepp, 2021-02-03:
Hi all,

Keen to follow this hal. Have a few boards coming and might try to port some of my bluepill code to the rp to see how it goes (long term).

Shorter term, is there a blinky sample/example to getting started?. I suspect the rpi pico will be fairly popular, so would be good to have a guide helping people with the basics :)

tdittr, 2021-02-03:
Hi,

I'm not sure what the current best example is (and my board is still in the mail). But there are these projects:
- https://github.com/rp-rs/sample-project
- https://github.com/rp-rs/pico-blink-rs

Also there is a Matrix chat with a bunch of helpful folks: https://matrix.to/#/#rp-rs:matrix.org

Hope that helps.

AsafFisher, 2021-02-03:
I think that answers the question (:

closed 2021-02-03 by AsafFisher