rp-rs/rp-hal #19: Add a blinky example using the HAL

tdittr, 2021-02-07:
Can be implemented after the following parts have been implemented:
- [ ] GPIO (#4)
- [ ] Clocks (#6)
- [ ] Resets

Use cases:
- Starting point for fresh projects
- Getting a quick overview of the HAL

chinoto, 2021-06-20:
The checklist hasn't been updated since [main/rp2040-hal/examples/blinky.rs](https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/examples/blinky.rs) was added. Based on what I'm seeing in the example, GPIO seems done and I see a [resets.rs](https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/src/resets.rs) exists, is resetting handled on this line?
https://github.com/rp-rs/rp-hal/blob/a14cbb581907cd05208705b51b953ac773abb0d5/rp2040-hal/examples/blinky.rs#L23

I'm probably misunderstanding, but couldn't timing be handled like it is in the non-HAL example? And where does the 8e6 for AHB frequency come from? 
https://github.com/rp-rs/pico-blink-rs/blob/247df96b9089096140a9713ebb4d40bd369ef730/src/main.rs#L111
```rust
let mut delay = cortex_m::delay::Delay::new(cp.SYST, 8_000_000);
```

I'm looking forward to helping out so my friend and I can use the Picos we bought before checking Rust support 😅. I'm particularly interested in getting SPI, I2C, and possibly PWM/DMA implemented.

9names, 2021-07-30:
Apologies for not updating the checklist.
It was really used more as a guide of what to work on, and what we had started work on, but wasn't being updated on things being complete (since it was unclear how "complete" each section needed to be before being ticked off)
> is resetting handled on this line?

Hard to say with this old code, but the relevant code in the current codebase is
https://github.com/rp-rs/rp-hal/blob/711c0230b1923492bc9bd8c9614a41e92d2070cc/rp2040-hal/src/gpio/pin.rs#L786-L801
which is invoked here
https://github.com/rp-rs/rp-hal/blob/2d1086915c3c178985445edaf559dbbbe8f4b8d3/boards/pico/examples/pico_blinky.rs#L46-L51
Perhaps resetting isn't being handled at all in the base example?
We should probably check, and if it's not, fix that. It would be a good first issue for you to tackle if you are keen.

> where does the 8e6 for AHB 

At the time we wrote the first examples I wasn't sure what the default frequency would be, and I didn't want anyone to think that the delays would be accurate, so we went with cycles instead.
My experience says the the frequency will actually be 12Mhz, not 8Mhz, but that may be just leftover configuration from the USB bootloader after loading some code.
It would be nice to have function in the HAL that returns the best guess of what the effective clock frequency is based on the how the various registers are set - only used on startup, when you haven't set it explicitly.
If you've already init'd the clocks, you can use the frequency returned from that.
https://github.com/rp-rs/rp-hal/blob/2d1086915c3c178985445edaf559dbbbe8f4b8d3/boards/pico/examples/pico_blinky.rs#L43

> I'm looking forward to helping out

Help is always welcome!
I recommend you join our matrix channel if you do though, since it's a bit easier for quick discussion than the comments section here
https://app.element.io/#/room/#rp-rs:matrix.org