rp-rs/rp-hal #22: Current state?

julianoes, 2021-04-10:
Hi there,

I would be interested to help with this to learn more about Rust on embedded. However, it doesn't look like there is active progress here at the moment. Are you expectingg this project to take up activity again, or is it stalled for now?

And if I were to try to continue here. Which issue or PR should be the first thing to look into? I see there are two draft PRs. I would assume, they need more work to get merged? If so, what is missing?

Thanks :wave:

tdittr, 2021-04-10:
Hi,

thanks for reaching out. Yes most development has stalled a bit. If you would like you could complete my PR for the Clock-Tree: https://github.com/rp-rs/rp-hal/pull/18

I finished the wrappers for XOSC and PLL but since than I could not find time to continue. You could pick up where I left of or rewrite it to be simpler. If you want I could help with code review.

My initial idea was to pass around the ownership of the clock sources via structs but it turned out to be to complicated. So I would maybe remove all the code that is not part of the XOSC or PLL modules.

Just let me know if I can help you in any way :)

tdittr, 2021-04-24:
just so you know there is now an open PR for the XOsc peripheral by @Nic0w: https://github.com/rp-rs/rp-hal/pull/25

Nitori-, 2021-04-24:
I've got GPIOs mostly working, and there's that clocks PR that is also making good progress.

Once those two land, I think we can start hitting the many different peripherals. Some will likely be easier than others (e.g. SPI vs PIO).

Nic0w, 2021-04-25:
Hey there !
So yeah there's my PR on the XOSC and my end goal was to have UART working. So I've also been working on PLLs as well as Clocks, and of course UART. 
All of those things have been working for me for two days now \o/ so I'm quite excited. 
 
I'm most happy with the code for the PLLs and the UART; I could push those next, and also possibly clocks but it's a big chunk of code that would need some polishing and discussions. 
I've also been working on GPIOs and as @Nitori- I tried to keep thing simple by focusing mostly on SIO but I've also got basic function switching (needed for UART).

tdittr, 2021-04-25:
> I'm most happy with the code for the PLLs and the UART; I could push those next, and also possibly clocks but it's a big chunk of code that would need some polishing and discussions.

Feel free to open a draft PR for the PLL and UART. I would be happy to have a look at them :smiley:

Nic0w, 2021-04-25:
here you go for the PLL, UART is on its way as well :)

9names, 2021-05-09:
@julianoes: project is not stalled, but we have a relatively small number of contributors working on this when they find the time. And we all have too many projects ;)
Please comment on a tracking issue for a peripheral if you want to help out with it, to ensure no-one works over the top of someone else.
I recommend you join the rust embedded and rp-rs matrix channels if you're new to Embedded Rust, there are a lot more people there that can ask questions and bounce ideas off.
https://app.element.io/#/room/#rust-embedded:matrix.org
https://app.element.io/#/room/#rp-rs:matrix.org

julianoes, 2021-05-09:
@9names thanks. And the same is apparently true for me, too many projects going on! :)

It's nice to know that it's not stalled though, and I will do some reading through the things mentioned. And I'll join the matrix channels, thanks for the link.

closed 2021-05-09 by julianoes