rp-rs/rp-hal #35: Hardware Divider/Modulo support

9names, 2021-05-09:
See section 2.3.1.5 of the [reference manual](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf) for more info on this peripheral.

This is part of the SIO peripheral, and may require use of a mutex or the hardware spinlocks provided by the SIO to use safely. In the initial implementation, having it block until complete would be recommended.
You can also assume (for the initial implementation) that it's never used by interrupts or by the 2nd core.

Minimum deliverable should be standalone function(s) with examples that demonstrate that results match built-in / and % operators.

eutampieri, 2021-05-14:
I'd like to contribute to this issue but I'm new to embedded rust. Would this be a good first issue nonetheless?

9names, 2021-05-15:
I think it is a good first issue if you're interested in getting into Embedded Rust HALs.
If you're just starting out you might find it a bit intimidating, but please feel free to ask for help or advice if you need it.

eutampieri, 2021-05-15:
If I understand correctly it is a hardware accelerated division (or modulo), right?
I’ll then have to get the SDK and try to implement it.
How is the PR linked to this specific issue? Is it the SIO spin lock implementation?

9names, 2021-05-15:
The hardware is described in detail in the datasheet that is linked in the issue description.
We aren't using the SDK, you would be accessing the hardware through the PAC crate https://github.com/rp-rs/rp2040-pac.
This is already included as a [dependency of the HAL](https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/Cargo.toml), so you can clone this repo, add a new file to rp2040-hal/src/, and start writing the function.

9names, 2021-05-15:
The PR linked to this issue was related to splitting out the SIO peripheral into chunks.
Until then, it was used exclusively by the GPIO module.
The most common model for Embedded Rust HALs is that a single module will *own* a portion of the memory-mapped register space, as that eliminates issues around concurrent access to the registers.
The SIO is used for GPIO, divide/modulo, spinlocks, message passing. Those things don't really all belong under the GPIO module, hence the PR to allow other modules to own their own space.

9names, 2021-06-02:
Initial support was merged in PR https://github.com/rp-rs/rp-hal/pull/40
It could use more testing, and will need some work to make it safe across interrupts and multiple cores - if anyone is interested in working on these tasks please open a new issue / draft PR.

closed 2021-06-02 by 9names

eutampieri, 2021-06-02:
Yeah sorry, I had a look but now I’m rather busy…