rp-rs/rp-hal #41: UART example?

Pjottos, 2021-05-27:
It is unclear to me how to use the UART peripherals. Specifically:
- What does the frequency parameter mean when calling `UartPeripheral::enable`? I cannot find anything about this online, only baud rate is mentioned.
- What pins are used for the peripherals? The rp2040 datasheet mentions that multiple pins can be used.

For testing i hooked up what i thought were the default tx and rx pins for UART0 (GP0 and GP1 respectively) to the rx and tx pins on a raspberry pi 3. Then i ran picocom on it with the same paramers as i use on the pi pico, baudrate 9600, data bits 8, stop bits 1, and parity none. I tried with frequencies 1, 1000 and 100000. Sometimes i would get some bogus output but most of the time there was nothing.

I'm not exactly experienced with embedded development so if i'm missing something obvious, do tell me.

9names, 2021-06-02:
> What does the frequency parameter mean

In any processor, different parts of the circuit take in a clock from one of a number of sources and divide it up to get the frequency they need to operate correctly. 
The clock they take this from will usually be part of a [clock tree](https://en.wikipedia.org/wiki/Clock_signal#Distribution)

In the case of a UART peripheral, the input clock is some multiple of the target baud rate to allow for oversampling the signal and other processing. On the Pico in particular it needs to be at least 16x the frequency of the baud rate you're targeting - which probably explains why you're getting gibberish on the Rpi3 when you aren't setting it high enough.
You want it to either be a multiple of the target baud or significantly greater, otherwise the actual baud rate will be too far from the desired rate to work correctly.
The [pico datasheet](https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf) describes this in more detail.
If you look at figure 59 on page 437 of the datasheet, you'll see the source clock signal listed as as UARTCLK.
The requirements for the UART peripheral clock input are described in much greater detail on page 439, section 4.2.3.2. _UART operation_.

> What pins are used for the peripherals

Rp2040 has a pin mux (multiplexer) that allows pins to connect to different peripherals.
Permitted mappings are also listed in the datasheet on page 13: 1.4.3 _GPIO Functions_

Pjottos, 2021-06-04:
Thank you, that clarifies it.

closed 2021-06-04 by Pjottos

Szpadel, 2021-07-25:
Can you share working example? I have issues where `UartPeripheral::enable` hangs board and I have no idea what I'm doing wrong

9names, 2021-07-25:
Please open a new issue rather than commenting on an already closed one.