rp-rs/rp-hal #62: UART example

Szpadel, 2021-07-25:
Hi,

Can anyone provide example for setting up UART?
I'm having issue that I located (using led) to be in `UartPeripheral::enable` (code probably panics or hangs the cpu)
I located that it cannot be invalid frequency (as I didn't unwrap error) and I didn't found anything obvious inside that could cause that.

Thanks

Szpadel, 2021-07-25:
I debugged that it hangs in this loop:
https://github.com/rp-rs/rp-hal/blob/148cc2b8ea83cfb035ff6abd76dbe43e6b7c1bc8/rp2040-hal/src/resets.rs#L19

Any hints why it might not go from reset?

leo60228, 2021-07-27:
I haven't tested this myself, but did you call `init_clocks_and_plls`? That's a pretty simple thing, but comparing the rp-hal and pico-sdk code, that's the only thing I could think of...

pico-sdk has a check to make sure clk_peri is enabled before initializing the UART.

PR #73 (2021-08-08): Add UART example + core::fmt support to UART
HAL UART code did not have an example, and the documentation did not have sample code either.
This PR adds both of these.

Resolves #62
closed 2021-08-08 by 9names