rp-rs/rp-hal #75: Issues when setting bits in registers

thejpster, 2021-08-08:
In `pwm.rs` I saw:

```
pad.gpio[self.pin].write(|w| w.ie().set_bit());
pad.gpio[self.pin].write(|w| w.od().clear_bit());
```

My understanding of `write` is that it does:

```
let w = register.defaults();
let w = closure(w);
register.set_bits(w);
```

So the second line (clearing `od`) will undo the effects of the first. This should probably be:

```
pad.gpio[self.pin].modify(|_r, w| {
    w.ie.set_bit();
    w.od.clear_bit();
    w
});
```

thejpster, 2021-08-08:
The default value is `0x56` or `0b01010110` which is bits 1, 2, 4 and 6. This is equal to:

* OD = 0 = Output Enabled
* IE = 1 = Input Enabled
* DRIVE = 0b01 = 4mA
* PUE = 0 = Pull-Up Disabled
* PDE = 1 = Pull-Down Enabled
* SCHMITT = 1 = Schmitt Input Enabled
* SLEWFAST = 0 = Slow GPIO Mode

I think this means the two lines I highlighted above are at least /harmless/, as they reset the pin to defaults, and then change one specific bit to a value that it already has in the default.

9names, 2021-08-08:
I mentioned this in matrix but didn't remember to make an issue for it.
This has been noted in a review comment on the PWM rework PR, but it's good to have a reminder that it's not done yet.
https://github.com/rp-rs/rp-hal/pull/61/commits/5b7265e711772668f8ec09845f327b32d9b1faee#r684648303

9names, 2021-08-14:
Fixed in #80

closed 2021-08-14 by 9names