rp-rs/rp-hal #92: Need to update references to rp2040-boot2-rs

taunusflieger, 2021-08-21:
Currently for boards other than the Pico, rp2040-boot2-rs is referenced by a specific commit as no released version has the board specific second stage boot loader. This should be changed once the next release of rp2040-boot2-rs is available.

thejpster, 2021-11-25:
I think we're good now - rp2040-boot2 0.2.0 came out a while ago?

9names, 2021-12-03:
Yep, we're good, going to close this issue.
```bash
rg rp2040-boot2
rp2040-hal/Cargo.toml
42:rp2040-boot2 = "0.2"

boards/feather_rp2040/Cargo.toml
21:rp2040-boot2 = "0.2"

boards/qt_py_rp2040/Cargo.toml
21:rp2040-boot2 = "0.2"

boards/pro_micro_rp2040/Cargo.toml
24:rp2040-boot2 = "0.2"

boards/itsy_bitsy_rp2040/Cargo.toml
21:rp2040-boot2 = "0.2"

boards/pico_lipo_16mb/Cargo.toml
22:rp2040-boot2 = "0.2"

boards/pico_explorer/Cargo.toml
30:rp2040-boot2 = "0.2"

boards/pico/Cargo.toml
39:rp2040-boot2 = "0.2"
```

closed 2021-12-03 by 9names