rp-rs/rp-hal #145: BSPs should export boot2

9names, 2021-09-30:
We're currently manually specifying which boot2 to use inside user/example code.
If the BSP exported a bootloader symbol, BSP examples could be generic.
Example
```rust
// alias for the bsp so we don't have to change as much code when changing targets
use feather_rp2040 as target_bsp;
#[link_section = ".boot2"]
#[used]
pub static BOOT2: [u8; 256] = target_bsp::BOOT_LOADER;
```
instead of a different
```rust
pub static BOOT2: [u8; 256] = rp2040_boot2::BOOT_LOADER_GD25Q64CS;
```
for each board

9names, 2021-09-30:
An alternative approach was suggested by adamgreig in matrix chat:
A default feature, that when enabled makes the BSPs link rp2040_boot2::BOOT_LOADER for you.
Pros:
- No extra boilerplate in every project, completely invisible to the user.

Cons:
- Being invisible means that if it stops working for whatever reason, it's more painful to fix
- Default features do not spark joy

jannic, 2021-10-01:
One possible disadvantage (of both approaches):
The build tools necessary to build the bootloader (arm-none-eabi-gcc) may not be easily available everywhere. (Windows?) 
In that case, one can easily include a precompiled boot2, instead:
```
#[link_section = ".boot2"]
#[used]
pub static BOOT2: [u8; 256] = *include_bytes!("boot2.bin");
```
But if `rp2040-boot2-rs` is a dependency of the BSP, one also needs to fork the BSP and remove that dependency.

(Please note that there is already a suggestion to change the boot2 crate so the assembler does not run automatically at build time: https://github.com/rp-rs/rp2040-boot2-rs/pull/2. With that change, my point would be void.)

thejpster, 2021-12-04:
Nearly there!

PR #153 (2021-12-04): Make BSPs provide boot2 as a default feature
Provide definition for `BOOT2` inside of each BSP hidden behind the `boot2` feature flag, and make `boot2` a default feature of the BSPs
This moves
```rust
#[cfg(feature = "boot2")]
#[link_section = ".boot2"]
#[used]
pub static BOOT2: [u8; 256] = rp2040_boot2::BOOT_LOADER_W25Q080;
```
from the BSP examples into the BSP library.
Users of the BSP then have one less thing to paste/change in their own code.

While I was already changing this code I also updated the HAL and all BSP crates to use the latest git commit for rp2040-boot2.

Closes #145

Might [... cut]
closed 2021-12-04 by 9names