rp-rs/rp-hal #156: memory.x only included by accident

jonathanpallant, 2021-10-05:
memory.x is only included by an `INCLUDE memory.x` line in the cortex-m-rt `link.x` file. Because we don't copy the file to ./target, the file is effectively only included by accident and only if you happen to be in the root of the git repo when you do the build.

Also, the sections are out of order which breaks https://github.com/raspberrypi/picotool/issues/39

SuperCuber, 2021-11-17:
Might be related: I'm trying to build an example and I get this error
```
error: linking with `rust-lld` failed: exit code: 1
<lots of arguments>
\target\thumbv6m-none-eabi\debug\build\cortex-m-rt-eb0715ff362a68a0\o
ut\link.x:23: cannot find linker script memory.x
          >>> INCLUDE memory.x
          >>>         ^
```

9names, 2021-11-17:
Might be. I'm having trouble reproducing it or @jonathanpallant's issue with it only building from the root of the repo.
I just tried doing clean builds of blinky on linux and windows
- in the root
- in the rp2040-hal directory
- in the rp2040-hal/examples directory
And builds of pico_blinky
- in the boards directory
- in the boards/pico/
- in the boards/pico/examples directory
Successfully built and linked every time.

Can you provide any more information on OS, how you're building, rust version, etc?

SuperCuber, 2021-11-17:
Windows 10, `cargo 1.56.0 (4ed5d137b 2021-10-04)`

I copied `.cargo/config` to my project, and the `pico_blinky.rs` example to my `main.rs`, then ran `cargo run` from the root of the project.
dependencies mostly copied from `pico/Cargo.toml`:
```toml
[dependencies]
pico = { git = "https://github.com/rp-rs/rp-hal.git" }
cortex-m = "0.7.2"
rp2040-hal = "0.2"
cortex-m-rt = { version = "0.7"}
embedded-time = "0.12.0"
usb-device= "0.2.8"
usbd-serial = "0.1.1"
usbd-hid = "0.5.1"
panic-halt= "0.2.0"
embedded-hal ="0.2.5"
cortex-m-rtic = "0.6.0-alpha.5"
rp2040-boot2 = "0.2"
nb = "1.0"
i2c-pio = { git = "https://github.com/ithinuel/i2c-pio-rs", rev = "fb6167d02b7fbc46a83f344f5242823bcd16e271" }
```

9names, 2021-11-17:
oh, in that case you will need a copy of memory.x in your project folder.
if you just want to run an example from the HAL, use this command from inside your rp-hal folder:
```
cargo build --release --example blinky
```
If you want something you can copy the examples into, can I suggest looking at https://github.com/rp-rs/rp2040-project-template?
It's configured by default to use a debugger (a second Pico will do, if you have one), but you could strip out a bunch of the dependencies and use the .cargo/config from the hal with it if you want to program over the USB bootloader.

SuperCuber, 2021-11-17:
@9names thanks a lot, that works! Time to mess around with it :D