rp-rs/rp-hal #202: The entry point of a PIO program may not be its first instruction.

ithinuel, 2021-11-17:
Should there be a ways to skip this when the entry point is not the first instruction of the program (eg I2C)?
https://github.com/rp-rs/rp-hal/blob/9d56062e447957f7304c31cbe7153abfe70a4f28/rp2040-hal/src/pio.rs#L1476-L1484

thejpster, 2021-11-20:
Surely we always need a jump? But maybe we could take an offset to add to the program start (which is in the variable `offset` above).

ithinuel, 2021-11-20:
The resolution of that offset isn't trivial as on some program it may be the start of the program (so be the offset of the program in the memory) or be a label somewhere in the program itself (ie an `entry_point` or a `wrap_target`).
I haven't thought too much about that but maybe using an `Option<&Label>` (provided the Label is bound).