rp-rs/rp-hal #215: Release a new version

jannic, 2021-11-28:
There has not been a new release of rp2040-hal for several months. During that time, the crate improved significantly.
With a new release (0.3.0?), it would be much easier for newcomers to find those improvements.

As rp2040-hal depends on [pio](https://github.com/rp-rs/pio-rs), which was not yet released to crates.io, there are a few steps to be taken:

- [ ] publish pio to crates.io
- [ ] update rp-hal/rp2040-hal to depend on the released version of pio
- [ ] publish updated version of rp2040-hal

9names, 2021-12-01:
One more task so I don't forget:
- [ ] Update changelog before tagging the release

thejpster, 2021-12-18:
embassy-rs is blocking us doing a release, because rp-hal git depends on embassy-traits.

thejpster, 2021-12-18:
Asked about it in https://github.com/embassy-rs/embassy/issues/549

thejpster, 2021-12-18:
Doesn't it complain that embassy-traits isn't a feature any more?

thejpster, 2021-12-18:
Apparently not...

thejpster, 2021-12-18:
https://github.com/rp-rs/rp-hal/pull/242

thejpster, 2021-12-20:
Done!

closed 2021-12-20 by thejpster