rp-rs/rp-hal #236: Add ability to create/register an interrupt vector table for Multicore

9names, 2021-12-08:
We can now create an entry-point for core 1, but we currently have no infrastructure in the HAL for setting up an interrupt vector table for that core.

jonathanpallant, 2021-12-10:
How do we control which cores get which interrupts? Do all interrupts go to both cores, and it depends on which interrupts are enabled in each of the two NVICs?

osimmac, 2022-07-12:
https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf
"2.3.2. Interrupts
Each core is equipped with a standard ARM Nested Vectored Interrupt Controller (NVIC) which has 32 interrupt inputs.
Each NVIC has the same interrupts routed to it, with the exception of the GPIO interrupts: there is one GPIO interrupt per
bank, per core. These are completely independent, so e.g. core 0 can be interrupted by GPIO 0 in bank 0, and core 1 by
GPIO 1 in the same bank.
On RP2040, only the lower 26 IRQ signals are connected on the NVIC, and IRQs 26 to 31 are tied to zero (never firing).
The core can still be forced to enter the relevant interrupt handler by writing bits 26 to 31 in the NVIC ISPR register."

being able to set these upper IRQ signals for software based interrupts would be cool, currently unable to figure out how to use RTIC because of that. It requires setting software based interrupts and listing their vectors.

even if just core 0 had access to the software based interrupts, I think it would be a good start.

9names, 2022-07-12:
The classic RTIC move is to pick an unused hardware interrupt vector for your software task. So if you're not using USB, for example, you could use the USB vector.

> it sounds like the upper IRQ signals that are unused are always shared between cores

That is not how I interpreted it at all. The NVICs are not connected, you are writing to ISPR which can only be done on the current core and is only visible to the current core.

jonathanpallant, 2022-07-12:
Do we expose the unused top 5 interrupts in the `Interrupts` enum? Perhaps we should, just for RTIC purposes.

osimmac, 2022-07-12:
> The classic RTIC move is to pick an unused hardware interrupt vector for your software task. So if you're not using USB, for example, you could use the USB vector.
>
>
> That is not how I interpreted it [...]

im basing that off of this bit: "Each NVIC has the same interrupts routed to it, with the exception of the GPIO interrupts: there is one GPIO interrupt per bank, per core. These are completely independent, so e.g. core 0 can be interrupted by GPIO 0 in bank 0, and core 1 by GPIO 1 in the same bank."

to me, thats saying that each NVIC routes the interrupt the sames way, unless its a hardware GPIO interrupt.

i could be misunderstanding, i'm quite a new to rust and embedded rust...

jonathanpallant, 2022-07-12:
[image]

jonathanpallant, 2022-07-12:
Each interrupt source (e.g. UART0) is be routed to both NVICs, and either can chose to wake up their attached Cortex-M0+ core, or ignore the interrupt. The only different is SIO which has two interrupt outputs - one for Core 0 only and one for Core 1 only.

jannic, 2022-07-12:
> to me, thats saying that each NVIC routes the interrupt the same way, unless its a hardware GPIO interrupt.

Right, but "IRQs 26 to 31 are tied to zero (never firing)". So those interrupts are never triggered from outside the individual core. Each core has it's own NVIC, so if the interrupt is triggered by writing to the NVIC ISPR register, the other core doesn't know about it.

jannic, 2022-07-12:
@jonathanpallant I think the sentence "there is one GPIO interrupt per bank, per core" doesn't refer to the SIO, but to the GPIO interrupt enable registers. See datasheet section 2.19.3. Interrupts: "There are enable, status, and force registers for three interrupt destinations: proc 0, proc 1, and dormant_wake. For proc 0 the registers are enable (PROC0_INTE0), status (PROC0_INTS0), and force ( PROC0_INTF0)."
So you can select individually per core which GPIO pin triggers the `IO_IRQ_BANK0` interrupt.

The SIO interrupts are not for GPIO, but for the FIFOs, see section 2.3.1.4. Inter-processor FIFOs

jonathanpallant, 2022-07-12:
Quite possibly. I haven't looked at the data sheet in a while.

9names, 2024-03-03:
I think this is fully resolved at this point.
The vector table for core1 is the same as for core0, so the process of registering interrupts is the same as for single-core programs.
The "different vector table for each core" problem is resolved by #321 - you start with the same vector table and dynamically update it in RAM as required.

closed 2024-03-03 by 9names