rp-rs/rp-hal #241: SPI formats

AxelMontini, 2021-12-12:
As an electronics beginner, I wanted to drive some ws2812b's with an SPI pin, not realizing that the only supported format right now is the Motorola one.

I was able to modify the set_format function myself to (hardcode) the TI format in the FRF register.

I would really like a way to change it with the set_format function directly, so that my christmas tree can shine more engonomically :smile:

AxelMontini, 2021-12-12:
I might open a PR myself later, I first need to figure out exactly what other settings are needed for each format

jannic, 2021-12-12:
Improvements of the API are off course welcome.
But for driving ws2812b from an rp2040, I'd use the PIO peripheral. 
There are several examples in the repository, for example https://github.com/rp-rs/rp-hal/blob/main/boards/feather_rp2040/examples/feather_neopixel_rainbow.rs
The actual code to make PIO speak the ws2812 protocol is at https://github.com/ithinuel/ws2812-pio-rs

jannic, 2022-06-22:
@AxelMontini: Out of curiosity, why do you need TI format for controlling a ws2812b?

jannic, 2022-06-24:
(I think this comment is a little bit off-topic. But I don't know a better place to discuss this. Anybody just interested in implementing the other SPI frame types supported by RP2040 can skip this comment. It's all about ws2812.)

Does it really cause any issue? I guess you are worried about the area marked "Q" in the diagram.  ("Q is an undefined signal").

However, the line is labeled SSPRXD, so as I understand it it's meant to describe a signal coming from some other SPI device.
About the transmit line, the data sheet says for motorola format: "during idle periods [...] the transmit data line SSPTXD is arbitrarily forced LOW". So on the TX line, there would not be an undefined signal, but just some time where the signal is low. That would increase the length of the low-time of one bit in the order of up to 1000ns (depending on clock settings). I don't think that this causes the ws2812 to misunderstand the bit.

Whereas for TI mode, "SSPTXD is tristated whenever the PrimeCell SSP is idle". So, without a pull-down, the line could drift to higher voltages and a 1 might be detected.

I guess both will work just fine, and I didn't yet see a tutorial for using SPI with WS2812 LEDs specifying one or the other frame mode. However, this is all theory, so if you actually observed that TI mode worked better than motorola mode, that would be interesting.

NelsonAPenn, 2023-07-21:
I've created a draft PR for this item. I still have yet to test with a device and scope.

It makes some breaking interface changes to not only set_format, but also some of the init functions. How unstable is the API permitted to be at this time?

jannic, 2023-07-21:
From my side, breaking changes that improve the API are fine. Of course it's better if we can reach the same goal without breaking changes, but I don't want to create an ugly API for compatibility reasons.

NelsonAPenn, 2023-07-22:
PR has been updated to fix all examples and doctests

PR #654 (2023-08-08): [spi] Make SPI set_format accept frame format
Allow selecting Texas Instruments SSI and National Semiconductor Microwire frame formats. Modify interface to only configure clock phase and polarity when applicable (if frame format is Motorola SPI).

Additional notes:

- resolves #241 
- contains breaking changes to both `set_format` and SPI init functions
- removes generic <M: Into<Mode>>
    - I could add a generic Mode to the `FrameFormat` enum, but I believe it adds more complexity than it is worth (`FrameFormat::MotorolaSpi(mode.into)` is concise enough
closed 2023-08-08 by jannic