rp-rs/rp-hal #244: Move the BSPs to their own crates

thejpster, 2021-12-20:
Now that we have a bunch of BSPs and they are relatively stable, I'd like to propose kicking them out of the rp-hal repository and giving each of them their own repository.

Advantages:

* It's much easier to tag and manage BSP releases, because the entire repo gets published as one package on crates.io

Disadvantages:

* A change to every BSP now requires 9 pull-requests instead of one.

thejpster, 2021-12-20:
Most of the BSP crates are publishable, except for `pico` which is using the unpublished embassy-traits package. We'll need to remove that.

```console
➜  boards git:(main) for i in *; do pushd $i; cargo publish --dry-run; popd; done 

~/Documents/rp-rs/rp-hal/boards/adafruit_macropad
    Updating crates.io index
   Packaging adafruit_macropad v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/adafruit_macropad)
   Verifying adafruit_macropad v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/adafruit_macropad)
  Downloaded rp2040-hal v0.3.0
  Downloaded 1 crate (109.8 KB) in 1.25s
    Finished dev [unoptimized + debuginfo] target(s) in 16.10s
   Uploading adafruit_macropad v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/adafruit_macropad)
warning: aborting upload due to dry run
~/Documents/rp-rs/rp-hal/boards

~/Documents/rp-rs/rp-hal/boards/feather_rp2040
    Updating crates.io index
   Packaging feather_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/feather_rp2040)
   Verifying feather_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/feather_rp2040)
    Finished dev [unoptimized + debuginfo] target(s) in 14.70s
   Uploading feather_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/feather_rp2040)
warning: aborting upload due to dry run
~/Documents/rp-rs/rp-hal/boards

~/Documents/rp-rs/rp-hal/boards/itsy_bitsy_rp2040
    Updating crates.io index
   Packaging itsy_bitsy_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/itsy_bitsy_rp2040)
   Verifying itsy_bitsy_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/itsy_bitsy_rp2040)
    Finished dev [unoptimized + debuginfo] target(s) in 14.64s
   Uploading itsy_bitsy_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/itsy_bitsy_rp2040)
warning: aborting upload due to dry run
~/Documents/rp-rs/rp-hal/boards

~/Documents/rp-rs/rp-hal/boards/kb2040
    Updating crates.io index
   Packaging kb2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/kb2040)
   Verifying kb2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/kb2040)
    Finished dev [unoptimized + debuginfo] target(s) in 14.60s
   Uploading kb2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/kb2040)
warning: aborting upload due to dry run
~/Documents/rp-rs/rp-hal/boards

~/Documents/rp-rs/rp-hal/boards/pico
    Updating crates.io index
error: all dependencies must have a version specified when publishing.
dependency `embassy` does not specify a version
Note: The published dependency will use the version from crates.io,
the `git` specification will be removed from the dependency declaration.
~/Documents/rp-rs/rp-hal/boards

~/Documents/rp-rs/rp-hal/boards/pico_explorer
    Updating crates.io index
   Packaging pico_explorer v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pico_explorer)
   Verifying pico_explorer v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pico_explorer)
    Finished dev [unoptimized + debuginfo] target(s) in 18.37s
   Uploading pico_explorer v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pico_explorer)
warning: aborting upload due to dry run
~/Documents/rp-rs/rp-hal/boards

~/Documents/rp-rs/rp-hal/boards/pico_lipo_16mb
    Updating crates.io index
   Packaging pico_lipo_16_mb v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pico_lipo_16mb)
   Verifying pico_lipo_16_mb v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pico_lipo_16mb)
    Finished dev [unoptimized + debuginfo] target(s) in 15.25s
   Uploading pico_lipo_16_mb v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pico_lipo_16mb)
warning: aborting upload due to dry run
~/Documents/rp-rs/rp-hal/boards

~/Documents/rp-rs/rp-hal/boards/pro_micro_rp2040
    Updating crates.io index
   Packaging pro_micro_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pro_micro_rp2040)
   Verifying pro_micro_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pro_micro_rp2040)
    Finished dev [unoptimized + debuginfo] target(s) in 14.97s
   Uploading pro_micro_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/pro_micro_rp2040)
warning: aborting upload due to dry run
~/Documents/rp-rs/rp-hal/boards

~/Documents/rp-rs/rp-hal/boards/qt_py_rp2040
    Updating crates.io index
   Packaging qt_py_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/qt_py_rp2040)
   Verifying qt_py_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/qt_py_rp2040)
    Finished dev [unoptimized + debuginfo] target(s) in 15.89s
   Uploading qt_py_rp2040 v0.1.0 (/home/jonathan/Documents/rp-rs/rp-hal/boards/qt_py_rp2040)
warning: aborting upload due to dry run
~/Documents/rp-rs/rp-hal/boards
```

ithinuel, 2021-12-20:
> except for pico which is using the unpublished embassy-traits package. We'll need to remove that.

Is it preventing publication?
It's behind a feature flag so I hoped it wouldn't be an issue.
Same for embassy-traits implementation in rp2040-hal.

9names, 2021-12-20:
I assume we will still build some BSPs as part of our CI to help catch unintended breakage?
Pro: a change to the HAL doesn't require N * number of BSP number of changes
Con: BSPs *will* get out of date.

The atsamd folks have a lot of BSPs and are trying to move to support only a few from the HAL team,  and let the extended community keep the others up to date.
Personally I only really care about Pico - I only have a couple of non-pico boards, and the only other BSP I care too much about isn't in the main repo anyway.

Can we have the BSPs in a single repo without a workspace, or will cargo not handle that?

thejpster, 2021-12-21:
> > except for pico which is using the unpublished embassy-traits package. We'll need to remove that.
> 
> Is it preventing publication? It's behind a feature flag so I hoped it wouldn't be an issue. Same for embassy-traits implementation in rp2040-hal.

You cannot cargo publish anything to crates.io with a git dependency, behind a feature flag or not. As embassy are not publishing embassy-traits any time soon, is is a blocker. The feature has already been removed from rp2040-hal.

thejpster, 2021-12-21:
> Can we have the BSPs in a single repo without a workspace, or will cargo not handle that?

cargo is fine with that. My only issue is tagging the repo and making it clear which tag each package was built from. With the packages in the repo that's probably manageable. With ten, it gets messy.

thejpster, 2021-12-21:
> I assume we will still build some BSPs as part of our CI to help catch unintended breakage?

BSPs in other repos will be using a pinned version of the HAL, so they shouldn't break. Yes we should test our API extensively so that we know if we've made a breaking change, but I would argue that we should do that in examples, not with an in-house BSP.

jannic, 2022-07-10:
With the huge number of BSPs, what about a middle-ground solution: Move the BSPs out of the rp-hal repo, into a single rp-bsps repo?
That'll still allow for rp-hal to do breaking changes without immediately updating all the BSPs, but avoids the problem that updating the BSPs needs gazillions of individual PRs.

9names, 2022-07-11:
I think this is the best way forward.
It cleans up the history + issue + PR queue for rp2040-hal, allows meaningful tags on rp2040-hal, and allows bulk updates to maintain BSPs.
We'll want to wait until #361 is complete to do this to avoid making more work for @ithinuel.

jannic, 2022-12-11:
This was implemented with #517

closed 2022-12-11 by jannic