rp-rs/rp-hal #246: Add extra humans to our crates, and fix team name

thejpster, 2021-12-23:
Only named humans can 'own' a crate, and hence have permissions to add or remove other owners.

'Team' membership merely grants publish and yank rights.

I propose we ensure that we always have at least two active humans listed against any crate, so they can add or remove teams as required in the future.

We also want to switch from `github:rp-rs:rp-rs` to `github:rp-rs:admins`.

These are the crates:

| Name | Human Owner | Team | Fixed |
|-------|---------------|-----|------|
| rp2040-pac | @eolder | rp-rs | ❌ | 
| rp2040-hal | @eolder | rp-rs | ❌ | 
| pio | @thejpster @Luminarys | admins | ✅ |
| rp2040-boot2 | @thejpster @jonathanpallant | admins | ❌ |
| rp-pico | @thejpster @jannic  | admins |  ✅ |

thejpster, 2021-12-31:
Pinging @eolder for comment

eolder, 2022-01-17:
Updated!

thejpster, 2022-01-18:
I removed jonathanpallant from boot2 and invited eolder. Nearly done!

thejpster, 2022-01-18:
Although I suppose there are the BSPs to consider

9names, 2022-03-06:
pio-proc and pio-parser are presently owned exclusively by @thejpster.
the admin team should get added to the owners on crates.io, too.

jonathanpallant, 2022-03-07:
Ah, good catch

thejpster, 2022-03-07:
Team added. I need a human to add too.

jannic, 2022-12-18:
There are still a few crates without a second human owner.

Also, I just noticed that I was the only owner for some BSP crates which where first uploaded by my when during the recent HAL+BSP releases. ([pimoroni-servo2040](https://crates.io/crates/pimoroni-servo2040), [waveshare-rp2040-lcd-0-96](https://crates.io/crates/waveshare-rp2040-lcd-0-96), [waveshare-rp2040-zero](https://crates.io/crates/waveshare-rp2040-zero))

I added the team to those crates. A second human is still missing.