rp-rs/rp-hal #250: Example pico_uart_irq_echo doesn't build

kolbma, 2021-12-30:
```
$ cargo build --release --example pico_uart_irq_echo
   Compiling rp-pico v0.3.0 ($HOME/temp/rp-hal/boards/rp-pico)
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.0.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.1.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.2.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.3.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.4.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.5.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.6.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.7.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.8.rcgu.o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a.pico_uart_irq_echo.e9e8dce0-cgu.9.rcgu.o" "--as-needed" "-L" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps" "-L" "$HOME/temp/rp-hal/target/release/deps" "-L" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/build/cortex-m-c99fffa40c11ec19/out" "-L" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/build/cortex-m-rt-2e3ab45dc71e1fd0/out" "-L" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/build/rp2040-pac-ddce79025e820932/out" "-L" "$HOME/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib" "-Bstatic" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/librp_pico-5ce06026e5edcc83.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/librp2040_boot2-ee58c2cef7b733a2.rlib" "--start-group" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libpanic_halt-da3a3e834882bb18.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/librp2040_hal-fc3105ede6730ae3.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libitertools-a52e9555ba3fdccb.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libeither-447541b88e797690.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/librand_core-95c2452ca9c99d15.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libcritical_section-ae582755c9d92a53.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libcfg_if-03adb85941389420.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libbare_metal-bfd7c87c0d3c9546.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libusb_device-0acff11a572f95a0.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libpio-c0b60f9b3cb27279.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnum_enum-cf675fc1e640c4f5.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libarrayvec-048c62ceaa2805cc.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/librp2040_pac-eb9a9e95f1e2ab95.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libcortex_m_rt-bc1b2d6e6ad55e36.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libcortex_m-7417ab55ee84e5f1.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libvolatile_register-f9d76bc3d5b612a6.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libvcell-1d660ee0d649a29c.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libbare_metal-15601a132a91655d.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libembedded_time-7c6ad800f94e40c7.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnum-b83afbb674de3139.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnum_iter-60df85c83bdf907d.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnum_rational-1af9399501e6fc86.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnum_integer-27148003c2c589a0.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnum_complex-342d588eba5c516e.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnum_traits-6b08ffd1e0f97dfc.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libembedded_hal-a96b34d4d4d619aa.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libvoid-41759aaa65ebdf70.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnb-954afe5ff02ba6df.rlib" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/deps/libnb-7c5642b570d45069.rlib" "$HOME/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/librustc_std_workspace_core-fe19048104d2a1bc.rlib" "$HOME/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/libcore-e92820fdfc0bef84.rlib" "--end-group" "$HOME/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib/libcompiler_builtins-dd984310dec02354.rlib" "-Bdynamic" "--eh-frame-hdr" "-znoexecstack" "-L" "$HOME/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/thumbv6m-none-eabi/lib" "-o" "$HOME/temp/rp-hal/target/thumbv6m-none-eabi/release/examples/pico_uart_irq_echo-7f4716ed3ccd6e1a" "--gc-sections" "-O1" "--nmagic" "-Tlink.x"
  = note: rust-lld: error: section '.boot2' will not fit in region 'BOOT2': overflowed by 256 bytes
          rust-lld: error: section '.boot2' will not fit in region 'BOOT2': overflowed by 256 bytes
          rust-lld: error: section '.boot2' will not fit in region 'BOOT2': overflowed by 256 bytes
          

error: could not compile `rp-pico` due to previous error
```

9names, 2021-12-31:
There are 3 BSP examples that still have BOOT2 declarations in them, which cause this issue now that #153 is merged and a BOOT2 is provided by the BSP via default feature.
I assume the pico ones were introduced since @thejpster wrote them originally as HAL examples.

I don't think we need 3 issues specifically to deal with this problem, 1 PR that removes those declarations will resolve this - #251 has the same symptoms and solution as this issue.

```console
~/rp-hal $ rg BOOT2 |grep examples
boards/adafruit-kb2040/examples/adafruit_kb2040_rainbow.rs:pub static BOOT2: [u8; 256] = rp2040_boot2::BOOT_LOADER_W25Q080;
boards/rp-pico/examples/pico_uart_irq_buffer.rs:pub static BOOT2: [u8; 256] = rp2040_boot2::BOOT_LOADER_W25Q080;
boards/rp-pico/examples/pico_uart_irq_echo.rs:pub static BOOT2: [u8; 256] = rp2040_boot2::BOOT_LOADER_W25Q080;
```

thejpster, 2021-12-31:
Huh. I wonder why CI didn't catch this

thejpster, 2021-12-31:
https://github.com/rp-rs/rp-hal/runs/4637772905?check_suite_focus=true

We only check examples, and only build 'all' - which apparently doesn't include examples.

Examples, along with everything else, should be built as debug and as release

thejpster, 2021-12-31:
'--all is a deprecated alias for --workspace'. Oh.

9names, 2022-01-03:
We should remove `--all` from the workflows then - I think it's only ever used with ``--workspace`` anyway.

I spent some time (way too long, actually) working on the workflows to get the error to show up.
It turns out that RUSTFLAGS: "-D warnings" masks the linker issues here as well, so even if we were doing `build` instead of `check` we would not have known it was broken.
Do you think we should remove that from the build_and_test workflow, or instead make a different workflow that just builds all the examples in release, debug, with e-h-1.0

PR #255 (2022-01-03): Remove static BOOT2 decl in BSP examples so they build
Fixes #250
Fixes #251
closed 2022-01-03 by thejpster