rp-rs/rp-hal #274: Add open drain I/O example to rp-pico board

BrunoWallner, 2022-01-23:
I am completely new to embedded programming and I want to read temperature data from a `dht11` sensor using [dht11-rs](https://github.com/plorefice/dht11-rs).
But it seems that this library needs access to a pin that both implements `embedded_hal::digital::v2::OutputPin` and `embedded_hal::digital::v2::InputPin`.

I do not know how a pin must must be initialized to meet these criteria and adding an example will help me and others who might have the same problem.

jannic, 2022-01-23:
As far as I know, rp-hal currently doesn't implement an open drain pin type.
But it's not too difficult to simulate one, at least in a way sufficient for dht11:
https://github.com/jannic/rp-hal/blob/dht11_example/rp2040-hal/examples/dht11.rs#L45-L69
(Not extensively tested, I just verified that I get a temperature reading at all.)

jannic, 2022-01-25:
@BrunoWallner, the ticket is still valid, I think. An example showing how to implement this would be a good thing. (Or even a ready-made implementation of such a pin mode in the HAL.)
When I find some time I'll provide a merge request from the branch I mentioned, but at least I have to update the comments, first. (The example was hastily copied from another one.)

jannic, 2022-01-28:
@BrunoWallner I updated the example to use the dht-sensor crate. It seems to be more reliable than the dht11 crate.
(https://github.com/plorefice/dht11-rs/issues/3#issuecomment-941293978)

closed 2022-01-31 by 9names