rp-rs/rp-hal #282: Should spinlock 31 be exposed to end users via the spinlock API?

9names, 2022-01-30:
I wonder if spinlock 31 should be removed here as it's used internally. I can't imagine people needing more than 30 spinlocks

_Originally posted by @VictorKoenders in https://github.com/rp-rs/rp-hal/pull/270#discussion_r790711369_

jannic, 2022-01-30:
I thought about this for a while.
On the one hand, in theory the same spinlock could be used by both user software and the HAL at the same time, if done carefully. And a bug would at worst cause a deadlock, so even without `unsafe`, there would not be a source of unsoundness.
But on the other hand, I don't see a good reason why one would want to do this, there are more then enough other spinlocks available. And in case some special use case really needs to access the same spinlock as the HAL, one could still write some unsafe code to access the register directly.

TL;DR: I agree with https://github.com/rp-rs/rp-hal/commit/4a540d041abf60c61ec93180946be995f2cfecac

Szpadel, 2022-01-31:
Why not document that and expose in a way that would require using unsafe function?

9names, 2022-01-31:
My rationale is quite selfish: I do not want to deal with bugs related to "action at a distance" when people use this spinlock and it dead-locks their critical_section. Unsafe+documentation is insufficient deterrent sometimes.
That, and I'm happy to delay dealing with problems of insufficient spinlocks until someone actually has a use-case that requires all 32.

9names, 2022-02-26:
I think it was open long enough, and can be reopened if anyone comes up with a good reason to do so.
Closing this now.

closed 2022-02-26 by 9names