rp-rs/rp-hal #284: `uart::Writer` isn't `Send`, but I think it could be

danielzfranklin, 2022-01-30:
This is based on a long conversation I had with adamgrieg on matrix, which didn't end with a definite conclusion. I'm not confident, but I think that `uart::Write` not being `Send` is a coincidence and not an actual requirement.

`UartPeripheral::split` gives the reader half `self.device`, and clones a new device for the writer.

```rust
    pub fn split(self) -> (Reader<UART0, P>, Writer<UART0, P>) {
        let reader = Reader {
            device: self.device,
            pins: self.pins,
            config: self.config,
            effective_baudrate: self.effective_baudrate,
        };
        // Safety: reader and writer will never write to the same address
        let device_copy = unsafe { &*UART0::ptr() };
        let writer = Writer {
            device: device_copy,
            device_marker: core::marker::PhantomData,
            pins: core::marker::PhantomData,
        };
       (reader, writer)
    }
```

`self.device` is the pac token that (in my limited understanding) proves "moral ownership", which the pac asserts is `Send`. In contrast, `device_copy` is the raw pointer to the `RegisterBlock`, which isn't `Send`.

```rust
pub struct UART0 {
    _marker: PhantomData<*const ()>,
}
unsafe impl Send for UART0 {}
impl UART0 {
    #[doc = r"Pointer to the register block"]
    pub const PTR: *const uart0::RegisterBlock = 0x4003_4000 as *const _;
    #[doc = r"Return the pointer to the register block"]
    #[inline(always)]
    pub const fn ptr() -> *const uart0::RegisterBlock {
        Self::PTR
    }
}
```

As a result the `Reader` is `Send` but the `Writer` isn't. I think this is just a coincidence of the implementation details. I don't know enough about how the registers work to know if the current implementation is sound. I do think this is equally sound:

```diff
     pub fn split(self) -> (Reader<UART0, P>, Writer<UART0, P>) {
         let reader = Reader {
             device: self.device,
             pins: self.pins,
             config: self.config,
             effective_baudrate: self.effective_baudrate,
         };
         // Safety: reader and writer will never write to the same address
-       let device_copy = unsafe { &*UART0::ptr() };
+       let device_copy = unsafe {
+           UART0 {
+               _marker: PhantomData,
+           }
+       };
         let writer = Writer {
             device: device_copy,
             device_marker: core::marker::PhantomData,
             pins: core::marker::PhantomData,
         };
         (reader, writer)
     }
```

marius-meissner, 2022-03-05:
@danielzfranklin Thanks for bringing up this topics and your analysis.
Unfortunately, I also lack the deep understanding to propose a reasonable solution here.

I would just like to note that this problem also prevents the use of (split) UART in RTIC.

What is your experience so far with the `Peripherals::steal().UART0` proposal? What do you think about just creating a PR for gaining some momentum?

danielzfranklin, 2022-03-05:
@marius-meissner 

When I wrote this I was experimenting with different embedded options. I've since settled on nrf52's with embassy, so I don't have any more practical experience.

I'm happy to make the PR (its a handful of lines of code), but I'm not in a position to do any sort of extensive real-world testing.

marius-meissner, 2022-03-05:
Alright, I replicated the changes in https://github.com/marius-meissner/rp-hal/tree/Issue-284 and currently testing the changes using UART for ESP8266 WiFi offloading in combination with RTIC.
Will keep this issue up2date regarding the results.

9names, 2022-05-15:
@marius-meissner: Is this issue resolved now that #314 is merged?

marius-meissner, 2022-05-16:
@9names Yes, this issue is resolved and can be closed.

closed 2022-05-26 by 9names