rp-rs/rp-hal #286: cannot synchronize PIOs with pio::StateMachine::synchronize_with()

astraw, 2022-02-01:
Hi, I'm having trouble synchronizing PIOs with the current implementation of `pio::StateMachine::synchronize_with()`. Bounds require this to be run on `Stopped` state machines but I am not able to start them synchronously and thus they run a little out of sync. I can get them sychronized if I relax the bounds requirements and create a `pio::Synchronize::start` method which starts all state machines simultaneously. I have done so in the branch https://github.com/astraw/rp-hal/tree/hack-synchronize-with (which also includes an example program `pio_synchronized.rs` showing this functioning). However, this particular implementation is troublesome in that it states the state machines without changing their type to `Running` from `Stopped`. I wonder if I am going about this wrong or if there is indeed not quite current API support for synchronizing PIO programs.

jannic, 2022-02-15:
Just tried it out myself, this does work:
https://github.com/jannic/rp-hal/blob/pio_synchronized/rp2040-hal/examples/pio_synchronized.rs

jannic, 2022-02-16:
> The `Synchronize` struct returned from `synchronize_with()` does restart the clocks when it is dropped so the idea of starting them from a stopped state doesn't seem so remote from what was intended.

It restarts the clock dividers, but it doesn't restart the state machine - two different things.

The state machines and their clocks are somewhat independent from each other. Of course, the Rust API could couple those concepts, but I think there are cases where it's useful to have them independent.
Eg. you may have two functions (like RX and TX) which can be enabled independently from each other, but when enabled, should both run synchronously to some clock signal.

However, an API to actually start multiple state machines at the same time would still be nice - independently of clock synchronization. Merging it into `struct Synchronize` would be difficult as that restarts the clock dividers on drop, unconditionally.

jannic, 2022-02-16:
I experimented a little bit, but I didn't find a nice interface to start multiple state machines at once, yet.
The issue is exactly that they all need to be set to `Running`, but everything I tried lead to some hell of generic parameters.

Starting exactly two state machines at once would be possible. But what about 3 or 4? Do we really need 4 different functions, like `start()`, `start2(other)`, `start3(other1, other2)` and `start4(other1, other2, other3)` to cover all cases?

astraw, 2022-02-16:
Yes, fear of generics hell is why I asked before attempting to proceed. I do think starting 2, 3, or 4 seems useful. Could a combinator approach like in `pio::Synchronize::and_with` be used?

jannic, 2022-02-16:
I got it working without that `irq wait` instruction, but I don't think the API is nice.
See latest commit in https://github.com/jannic/rp-hal/blob/pio_synchronized/rp2040-hal/examples/pio_synchronized.rs
But perhaps this is a starting point to find a better API?

astraw, 2022-02-17:
Yes, that is certainly better than my hack.

I wonder if the `Started` struct and the `Starting` marker could be replaced with (likely macro-generated) types that kept all the individual state machine markers <SMx> together? All possible combinations (SM1, SM2), ... (SM1, SM2, SM3, SM4) would be needed. While the implementation of such seems a bit complex and may face a dead end from generics hell, my hope would be that it would make actual usage more straightforward.

With the goal of eliminating the potential panic in `Started::take` could a marker for each PIO could also be used? It seems this could prevent the relevant failure mode at compile time.

To bikeshed, "thing.join()" to me would imply blocking while waiting for thing to finish so perhaps a better word here?

jannic, 2022-02-17:
I think I found a way to implement an API like this:
```
   let (sm0, sm1) = sm0.with(sm1).start().free();
```
If it works, it'll even allow to integrate the clock synchronization step:
```
   let (sm0, sm1) = sm0.with(sm1).sync_clock_dividers().start().free();
```
And, of course, this should also work with more than two state machines:
```
   let (sm0, sm1, sm2) = sm0.with(sm1).with(sm2).start().free();
```
Not yet finished, though. I hope I don't run into some blocker.

jannic, 2022-02-17:
Ok, I think the API is usable now:
```
    // Start both SMs at the same time
    let group = sm0.with(sm1).sync().start();
    // Stop both SMs at the same time
    let group = group.stop();
    // Start them again and extract the individual state machines
    let (sm1, sm2) = group.start().free();
    // Stop the two state machines separately
    let _sm1 = sm1.stop();
    let _sm2 = sm2.stop();
```
https://github.com/jannic/rp-hal/commit/9b625a573e0ad05c699845239ab3fdb2cae75fd1

PR #301 (2022-03-18): Allow to start multiple state machines in sync
This adds a bunch of boiler-plate structs to pio.rs, but is the easiest way I found to solve #286. Any idea for a better implementation is welcome.

The resulting API is fine though, as those types do not become visible:
```
    // Start both SMs at the same time
    let group = sm0.with(sm1).sync().start();

    // Stop both SMs at the same time
    let group = group.stop();

    // Start them again and extract the individual state machines
    let (sm1, sm2) = group.start().free();

    // Stop the two state machines separately
    let _sm1 = sm [... cut]
closed 2022-03-18 by 9names