rp-rs/rp-hal #297: Conflict over Sio between Multicore and other Peripherals

kevinastone, 2022-02-21:
I'm trying to spawn up a loop on Core1, but keep GPIO and other functions on Core 0.  `Multicore::new` takes a `&mut Sio`, but using other peripherals like GPIO create partial moves out of `Sio` causing a conflict with the borrow checker.

It seems that multicore only needs a mutable reference to `Sio.fifo`, can we change the calling convention to providing the fifo?

Example:

```
// let pac = [...]
let sio = Sio::new(pac.SIO);
let pins = rp_pico::Pins::new(
    pac.IO_BANK0,
    pac.PADS_BANK0,
    sio.gpio_bank0,
    &mut pac.RESETS,
);
let mut mc = multicore::Multicore::new(&mut pac.PSM, &mut pac.PPB, &mut sio);
```

```
<line> |     let mut mc = multicore::Multicore::new(&mut pac.PSM, &mut pac.PPB, &mut sio);
           |                                                                                                                              ^^^^^^^^ cannot borrow as mutable

error[E0382]: borrow of partially moved value: `sio`
```

jannic, 2022-02-21:
From a quick look at the code, I'd say you are right and it would be sufficient if `Multicore` took a reference to `Sio.fifo`.
However, this also works:
```
let sio = Sio::new(pac.SIO);
let mut mc = Multicore::new(&mut pac.PSM, &mut pac.PPB, &mut sio);
let cores = mc.cores();
let core1 = &mut cores[1];
let _ = core1.spawn(core1_task, unsafe { &mut CORE1_STACK.mem });
let pins = rp_pico::Pins::new(
    pac.IO_BANK0,
    pac.PADS_BANK0,
    sio.gpio_bank0,
    &mut pac.RESETS,
);
```
As long as the intermediate objects are not accessed further down, the compiler detects that they can all be dropped before the call to `Pins::new`. That way, the borrows are no longer active.

As the multicore API is rather new, there are probably many improvements possible. Any feedback is welcome!

kevinastone, 2022-02-22:
Great, that worked (and unblocked me).  I guess something to think about a further iteration on the multicore API.

closed 2022-02-22 by kevinastone