rp-rs/rp-hal #307: Error: "Memory segment 0x010000->0x010094 is outside of valid address range for device"

brandonros, 2022-03-02:
Cargo.toml:

```toml
[package]
name = "rp2040"
version = "0.0.1"
authors = ["Brandon Ros"]
edition = "2018"

[dependencies]
panic-halt = "0.2.0"
cortex-m = "0.7.4"
cortex-m-rt = "0.7.1"
embedded-hal = "0.2.7"
embedded-time = "0.12.1"
rp-pico = "0.2.0"

[[bin]]
name = "main"
path = "src/main.rs"
```

src/main.rs:

```rust
//! # Pico Blinky Example
//!
//! Blinks the LED on a Pico board.
//!
//! This will blink an LED attached to GP25, which is the pin the Pico uses for
//! the on-board LED.
//!
//! See the `Cargo.toml` file for Copyright and licence details.

#![no_std]
#![no_main]

// The macro for our start-up function
use cortex_m_rt::entry;

// GPIO traits
use embedded_hal::digital::v2::OutputPin;

// Time handling traits
use embedded_time::rate::*;

// Ensure we halt the program on panic (if we don't mention this crate it won't
// be linked)
use panic_halt as _;

// Pull in any important traits
use rp_pico::hal::prelude::*;

// A shorter alias for the Peripheral Access Crate, which provides low-level
// register access
use rp_pico::hal::pac;

// A shorter alias for the Hardware Abstraction Layer, which provides
// higher-level drivers.
use rp_pico::hal;

/// Entry point to our bare-metal application.
///
/// The `#[entry]` macro ensures the Cortex-M start-up code calls this function
/// as soon as all global variables are initialised.
///
/// The function configures the RP2040 peripherals, then blinks the LED in an
/// infinite loop.
#[entry]
fn main() -> ! {
    // Grab our singleton objects
    let mut pac = pac::Peripherals::take().unwrap();
    let core = pac::CorePeripherals::take().unwrap();

    // Set up the watchdog driver - needed by the clock setup code
    let mut watchdog = hal::Watchdog::new(pac.WATCHDOG);

    // Configure the clocks
    //
    // The default is to generate a 125 MHz system clock
    let clocks = hal::clocks::init_clocks_and_plls(
        rp_pico::XOSC_CRYSTAL_FREQ,
        pac.XOSC,
        pac.CLOCKS,
        pac.PLL_SYS,
        pac.PLL_USB,
        &mut pac.RESETS,
        &mut watchdog,
    )
    .ok()
    .unwrap();

    // The delay object lets us wait for specified amounts of time (in
    // milliseconds)
    let mut delay = cortex_m::delay::Delay::new(core.SYST, clocks.system_clock.freq().integer());

    // The single-cycle I/O block controls our GPIO pins
    let sio = hal::Sio::new(pac.SIO);

    // Set the pins up according to their function on this particular board
    let pins = rp_pico::Pins::new(
        pac.IO_BANK0,
        pac.PADS_BANK0,
        sio.gpio_bank0,
        &mut pac.RESETS,
    );

    // Set the LED to be an output
    let mut led_pin = pins.led.into_push_pull_output();

    // Blink the LED at 1 Hz
    loop {
        led_pin.set_high().unwrap();
        delay.delay_ms(500);
        led_pin.set_low().unwrap();
        delay.delay_ms(500);
    }
}

// End of file
```

.cargo/config:

```
[build]
target = "thumbv6m-none-eabi"

[target.thumbv6m-none-eabi]
runner = "elf2uf2-rs -d"
```

Output:

```
Brandons-MacBook-Air:rp2040 brandonros 2022-03-02 17:38:17 $ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
     Running `elf2uf2-rs -d target/thumbv6m-none-eabi/debug/main`
Found pico uf2 disk /Volumes/RPI-RP2
Error: "Memory segment 0x010000->0x010094 is outside of valid address range for device"
```

brandonros, 2022-03-02:
I most likely shouldn't have opened this here and instead should track it here I am guessing: https://github.com/JoNil/elf2uf2-rs/issues/9

9names, 2022-03-04:
> most likely shouldn't have opened this here and instead should track it here

I think you opened it in the correct place. I think your project is missing a couple of the essential bits to build properly.

Your .cargo/config needs to have 
```
rustflags = [
"-C", "link-arg=-Tlink.x",
]
```
you need a copy of `memory.x` from the rp-hal placed in your project folder, and also a build.rs file that will copy it to the correct place - otherwise it won't link correctly, and bits of your code will end up in the wrong place or will be completely missing from the built binary.

There's a github template with all of this already set up which is a much easier place to start.
https://github.com/rp-rs/rp2040-project-template
You'll need to replace the runner in .cargo/config to use it without a debugger, but otherwise it should "Just Work".

mike-kfed, 2022-04-01:
Just a heads up for people running into this: Even when using the rp2040 project template it can happen when you have `RUSTFLAGS` environment variable set, because it has precedence over the `.cargo/config.toml` file. I ran into this on my CI/CD scripts which have custom RUSTFLAGS to forbid bad coding practices (and didn't include the custom linker options). 
tldr: make sure this env variable is unset to successfully build a raspberry pico uf2 image ;)

9names, 2022-05-15:
@brandonros: did you get this working?

closed 2022-05-15 by brandonros