rp-rs/rp-hal #322: Add other ADC modes: free-running, FIFO

9names, 2022-04-02:
The current ADC implementation in the HAL only provides support for blocking one-shot samples.

Tasks:
- Provide an API to configure the ADC into free-running mode.
- Provide an API to configure the ADC sample FIFO, pacing timer, round-robin sampling and right shift options.
- Add an example showing off each mode added

9names, 2022-04-02:
[C SDK reference for configuring the ADC](https://github.com/raspberrypi/pico-sdk/blob/2062372d203b372849d573f252cf7c6dc2800c0a/src/rp2_common/hardware_adc/include/hardware/adc.h#L178)
[C example using polling in continuous mode](https://github.com/raspberrypi/pico-examples/blob/2fb08a028f886624ebc9c938775e054860549159/adc/adc_console/adc_console.c#L17)

9names, 2022-04-02:
There is no example in the C SDK for capturing multiple channels or handling the samples in the interrupt handler, though they do have one that [uses DMA to capture the data](https://github.com/raspberrypi/pico-examples/blob/2fb08a028f886624ebc9c938775e054860549159/adc/dma_capture/dma_capture.c#L74)

riskable, 2022-04-04:
Free-running mode support would be *fantastic*!  Very much looking forward to this feature.

For reference, "the plan" is to hook up an analog multiplexer to a single analog pin on the RP2040 and then hook up additional multiplexers to each channel of the one that's connected to the RP2040 (i.e. multiplexers through a multiplexer).  In free-running mode (without round-robin) this should--in theory--allow one to effectively bypass the RP2040's internal ADC MUX and replace it with an external one with the fastest performance possible...  No need to issue blocking calls that tell the RP2040 which ADC pin to read from and enable non-blocking reads (well, reads that don't require 96 cycles).

Hopefully that'll be possible :smile:

9names, 2023-07-12:
Resolved by #626 and #636

closed 2023-07-12 by 9names