rp-rs/rp-hal #327: rp-pico pico_usb_twitchy_mouse not working

m5p3nc3r, 2022-04-14:
Hi all

I am trying to build a custom USB HAL device for my rp-pico device.  When I try using the pico_usb_twitchy_mouse example, I can see the USB device through lsusb, but using hidviz I am unable to see any HID devices.

For reference
- My host machine is an M1 based Mac
- I am using probe-run

output from lsusb is:
```
Bus 002 Device 005: ID 16c0:27da 16c0 Twitchy Mousey  Serial: TEST
```

Regards
Matt

9names, 2022-04-14:
what is hidviz?

m5p3nc3r, 2022-04-14:
Its a tool for visualising HID interfaces - its free and very useful.

https://hidviz.org/

9names, 2022-04-14:
there is nothing on that page or on the git repo that indicates that hidviz supports macos.
on github there is an open issue for this: https://github.com/ondrejbudai/hidviz/issues/13
does it work for any other hid devices?

it doesn't even show up any devices on my linux system.

m5p3nc3r, 2022-04-14:
It works fine with all other devices - needs to be run as root if you don't have the correct permissions set though.

9names, 2022-04-14:
ah, i couldn't work out what the capture button was on the UI.
it seems to work just fine with that example for me though
[image]

m5p3nc3r, 2022-04-14:
Can I ask what your setup is?

9names, 2022-04-14:
sure. ubuntu 20.04 x86, kernel 5.16

m5p3nc3r, 2022-04-14:
Ok, will see if I can replicate on a Linux host then.  Thanks for your input.

9names, 2022-04-14:
i'm mostly surprised that it works at all, since it doesn't appear the that author tested on anything other than windows and linux

m5p3nc3r, 2022-04-14:
I love a bit of hacking!
It was relatively simple to get working on macos - just brew install qt5, cmake libusb and protobuf and it worked for me (with a slight update to the CMakeFiles) :-). Upstream hasn't been touched for a couple of years, so I'm assuming the author is not willing to support additional platforms.
I suppose I could add details of how to build to the upstream project.

9names, 2022-04-14:
willing is probably the wrong word. maybe they don't have any mac systems - i sure don't!
I think it would be nice of you to make a PR with updated docs if you've got it working.

m5p3nc3r, 2022-04-14:
Fair point - I wouldn't have one if it wasn't a work one!  I'll see what I can do to create a PR to update the documentation.  It is a great tool.

0xa10, 2022-05-13:
Did you ever figure out the issue with the example? It doesn’t work for me either, appears like an issue in usbd-hid

9names, 2022-05-15:
do you get the same symptoms as @m5p3nc3r or are you seeing something else?
same OS, etc?

m5p3nc3r, 2022-05-15:
A quick update on this.  I can compile the app on macos fine, and if I plug it into a linux machine it gets the expected twichy mouse effect.  But it is not recognised on the mac as a HID device, and I don't see any movement of the pointer.

I believe that the mac is rejecting the HID descriptors, but I have not had time to dig any deeper into this yet.

jannic, 2022-05-15:
@m5p3nc3r could you try changing the device_class from `0xEF` to `0` in line 116 of `pico_usb_twitchy_mouse.rs`?

I don't know much about USB so I might be completely wrong, but the description of `0xEF` at https://www.usb.org/defined-class-codes#anchor_BaseClassFEh doesn't seem to fit for a mouse device, and my "real" mouse uses a device_class of 0.
Perhaps that's what makes your mac reject the device?

9names, 2022-05-16:
It looks like that was copied straight out of the Microsoft example which was for a USB webcam. Would be cool if that's the only thing that's making MacOS sad!

m5p3nc3r, 2022-05-16:
Hey @jannic - great spot, I changed the device_class to '0' and it is now working on my mac.

PR #339 (2022-05-16): Set device_class of twitchy_mouse example to 0
This fixes the mouse on Mac.

Closes #327
commit db1b5d36 (2022-05-16): Set device_class of twitchy_mouse example to 0

This fixes the mouse on Mac.

Closes #327
closed 2022-05-16 by 9names