rp-rs/rp-hal #335: rust-lld: error: duplicate symbol: __aeabi_idivmod

paddywwoof, 2022-05-05:
Just started looking at using rust with the pico 2040. Followed the instructions on the README but got the above error at the linking stage. I'm sure it's something obvious and stupid, rather than an issue.

Running on ubuntu ryzen 5 if that's relevant.

PS I had to `rustup target add thumbv6m-none-eabi` so maybe that should go in the instructions

9names, 2022-05-05:
Which instructions are you referring to? The rp2040-hal readme that step in the "Getting Started" section, maybe we should just link to that for each of the board readmes to avoid duplication.

As for the compiler error: Are you on the stable toolchain? Which version of rust?
We've had issues with the latest nightly but no reports of issues on stable yet.
If you are not on 1.60 yet, try running `rustup update` and building again.

paddywwoof, 2022-05-06:
@9names, Thanks so much for your quick response. That was exactly my problem.

I was following https://crates.io/crates/rp-pico (also listed here https://github.com/rp-rs/rp-hal/tree/main/boards/rp-pico which is where I clicked on the `Issues` tab to see if anyone else had the same error)

However, if I had started from here https://github.com/rp-rs/rp-hal then I would have completed the critical `rustup` and optional `cargo` steps missing from the other instructions. That's the trouble with following a google search - I wasn't starting from page0!

closed 2022-05-06 by paddywwoof