rp-rs/rp-hal #342: PWM with servo seems not working properly

lucazulian, 2022-05-23:
I'm trying to move a micro servo with Raspberry Pi Pico and a simple micro servo motor but with the following code, the servo is not precise as the experiments I made with an Arduino board. I think the main reason would be the frequency with which I move the servo but I cannot find a way to have a similar precision.

```rust
//! # PWM Gear Steering Servo
//!
//! PWM attached to GP1

#![no_std]
#![no_main]

use cortex_m::delay::Delay;
// The macro for our start-up function
use cortex_m_rt::entry;

// GPIO traits
use embedded_hal::PwmPin;

// Time handling traits
use embedded_time::rate::*;

// Ensure we halt the program on panic (if we don't mention this crate it won't
// be linked)
use panic_halt as _;

// Pull in any important traits
use rp_pico::hal::prelude::*;

// A shorter alias for the Peripheral Access Crate, which provides low-level
// register access
use rp_pico::hal::pac;

// A shorter alias for the Hardware Abstraction Layer, which provides
// higher-level drivers.
use rp_pico::hal;

// The minimum PWM value we want
const LOW: u16 = u16::MIN;

// The maximum PWM value we want
const HIGH: u16 = u16::MAX;

/// Entry point to our bare-metal application.
///
/// The `#[entry]` macro ensures the Cortex-M start-up code calls this function
/// as soon as all global variables are initialised.
///
/// The function configures the RP2040 peripherals, then moves the servo in an
/// infinite loop.
#[entry]
fn main() -> ! {
    // Grab our singleton objects
    let mut pac = pac::Peripherals::take().unwrap();
    let core = pac::CorePeripherals::take().unwrap();

    // Set up the watchdog driver - needed by the clock setup code
    let mut watchdog = hal::Watchdog::new(pac.WATCHDOG);

    // Configure the clocks
    //
    // The default is to generate a 125 MHz system clock
    let clocks = hal::clocks::init_clocks_and_plls(
        rp_pico::XOSC_CRYSTAL_FREQ,
        pac.XOSC,
        pac.CLOCKS,
        pac.PLL_SYS,
        pac.PLL_USB,
        &mut pac.RESETS,
        &mut watchdog,
    )
    .ok()
    .unwrap();

    // The single-cycle I/O block controls our GPIO pins
    let sio = hal::Sio::new(pac.SIO);

    // Set the pins up according to their function on this particular board
    let pins = rp_pico::Pins::new(pac.IO_BANK0, pac.PADS_BANK0, sio.gpio_bank0, &mut pac.RESETS);

    // The delay object lets us wait for specified amounts of time (in
    // milliseconds)
    let mut delay = cortex_m::delay::Delay::new(core.SYST, clocks.system_clock.freq().integer());

    // Init PWMs
    let mut pwm_slices = hal::pwm::Slices::new(pac.PWM, &mut pac.RESETS);

    // Configure PWM0
    let pwm = &mut pwm_slices.pwm0;
    pwm.set_div_int(2);
    pwm.set_div_frac(175u8);
    pwm.set_ph_correct();
    pwm.enable();

    // Output channel B on PWM0 to the gpio1 pin
    let channel = &mut pwm.channel_b;
    channel.output_to(pins.gpio1);

    // Infinite loop, moving motor from 0 to 180 and vice-versa
    loop {
        // 0 to 180
        delay.delay_us(1);
        channel.set_duty(HIGH);
        
        // 180 to 0
        delay.delay_us(1);
        channel.set_duty(LOW);
   }
}

// End of file
```
Is there a way to have a 50 Hz frequency with `set_div_int` and `set_div_frac`? Is there another way to move a micro servo motor?

sonnny, 2022-05-23:
I have some code snippets on the link below, I used Pico's PIO to accomplish the pwm needed for the servo. I translated various pwm I found on the Internet. You may need to adjust the pulse width, I found out that different servos need different adjustments (i.e. perfect center is 1500us, but I have several servos from different manufacturers that responds a little bit differently, some above 1500us to center, some below 1500us to center). As far as the 50hz, I hooked up a cheap logic analyzer from Amazon to experiment.

link --> [http://www.crus.in/codes/pio_demos.html](http://www.crus.in/codes/pio_demos.html) 

Sonny

lucazulian, 2022-05-24:
@sonnny, thank you so much; you made my day! Maybe I can open a PR with an example for the servo case. What do you think about it?

sonnny, 2022-05-25:
@lucazulian, you're very welcome, go ahead with a PR with an example, I think it's a good idea.

PR #346 (2022-05-26): Add Pico PWM micro servo example
Closes: #342
closed 2022-05-26 by 9names