rp-rs/rp-hal #348: Move external hardware examples out of rp-hal and into boards/rp-pico

9names, 2022-05-29:
The rp-hal examples should be small and specific to the RP2040 hardware.
We should move anything that's really an example of interfacing with external hardware somewhere else.
Moving them under the rp-pico examples would be a good first step in this process, I think.

There's only two examples that use external hardware that I can see:
- https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/examples/dht11.rs
- https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/examples/lcd_display.rs

#347 has already ported lcd_display to rp-pico (and added some nice diagrams) so we should be able to drop rp2040-hal/examples/lcd_display.rs if we accept this PR.

paddywwoof, 2022-05-30:
I agree with this but it probably needs thinking through.

I only have the pico board so any code I cobble together will be tested on that and `boards/rp-pico/examples/` seems logical. However it feels wrong to duplicate essentially the same code in each of the boards examples folders, with just the GPIO pins changed. Alternatively, spreading examples of interfacing with different hardware randomly across the different boards would be even worse as it would make searching for a specific one rather tedious.

jannic, 2022-05-30:
I agree that duplicating examples to each of the boards seems like a bad idea.
However it could be an option to just put them to the `rp-pico` example folder. As `rp-pico` is a kind of reference design for RP2040 boards, it makes sense to use it even for examples which are not `rp-pico` specific. It's also quite cheap and common.
So the distinction could be:
- `rp2040-hal/examples/` for examples specific to RP2040 internal hardware
- `boards/<board>/examples` for board-specific examples
- `boards/rp-pico/examples` for examples using external hardware.

This mixes two kinds of examples within `boards/rp-pico/examples`, but I guess that can be tolerated.

9names, 2022-05-31:
I should have fleshed out the rationale a bit better, but @jannic has correctly outlined the intention here.
It helps that Pico is basically a clean slate - other than the LED and USB there is basically no hardware to support.

paddywwoof, 2022-05-31:
Yes, that makes complete sense.

I suppose the next question is how much stuff should be in here as a "batteries included" repository. The vanilla Arduino sdk has [sixty or so built in](https://docs.arduino.cc/built-in-examples/)  but raspberrypi org seems to have broken them out into different repositories see [here](https://github.com/raspberrypi/pico-examples), [here](https://github.com/raspberrypi/pico-extras) and [here](https://github.com/raspberrypi/pico-micropython-examples), and [this](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf#%5B%7B%22num%22%3A283%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C115%2C841.89%2Cnull%5D) is nice.

Perhaps we should prevail on raspberrypi org to open a repository `pico-rust-examples'. They really ought to have someone on their team dedicated to using rust on their hardware by now!

thejpster, 2022-06-05:
I like the idea of an rp-rs-playground repo containing multiple individual binary crates. I also agree the examples in the rp-hal crate should be just enough to demonstrate how to use the HAL APIs and the BSP examples should be just enough to demonstrate how to use the BSP API.

9names, 2022-06-24:
After spending more time reviewing the DHT example, I think it's probably okay?
It's primarily an example of changing pin modes while talking to the hardware, so it's not *really* an external hardware example as much as a pin mode tutorial.

jannic, 2022-06-24:
> After spending more time reviewing the DHT example, I think it's probably okay? It's primarily an example of changing pin modes while talking to the hardware, so it's not _really_ an external hardware example as much as a pin mode tutorial.

It could probably be further reduced to some "how do I simulate an open-drain GPIO on RP2040" example, without relation to DHT11. But then, having _some_ practical motivation in an example is also nice.

9names, 2022-07-11:
> having some practical motivation in an example is also nice

Agreed! I'm against having hardware in one of the examples for no reason, but there is good reason in this instance.
Going to close this now as this issue has served its purpose.

closed 2022-07-11 by 9names