rp-rs/rp-hal #373: Support binary format  file

imrank03, 2022-06-29:
I have written a simple led blinking program, but if I flash the binary file on the `pico`, the led does not blink. Does `rp2040` support binary file?

ExplodingWaffle, 2022-06-29:
The pico’s bootloader takes UF2 files, but the binaries produced by rust are ELF. You have to convert it- I’d suggest having a closer look at the readme section on [programming](https://github.com/rp-rs/rp-hal#loading-a-uf2-over-usb) and using the cargo runner to do it for you.

9names, 2022-06-29:
@imrank03: please do not ask the same question in multiple venues at the same time.
This is disrespectful: you are telling us that everyone else's time has no value to you.

If you are asking questions on github, please make some effort to describe your problem.
Here's a breakdown of what would be a good issue based on the conversion on the embdedded-rust matrix channel:
- what you are trying to do: "I want flash binary format on the pico not executable format, because I'm writing a secure boot for rp2040, so i want binary file for signature"
- how you are trying to do it: "I'm building the blinky example from rp-hal, and running it through "rust-objcopy rp2040_blinky  -O binary rp2040_blinky.bin"
- what you have tried:  "my board blinks when I flash the elf file via probe-rs-cli, but when I try to flash rp2040_blinky.bin produced from the same file via probe-rs-cli it does nothing."

imrank03, 2022-06-30:
@9names: I don't mean to disrespect anyone. Sorry if you were disrespectful.

Yeah sure, if I'm asking on Github I'll describe the problem, thank you.

- I tried the latest probe-rs-cli, by doing this `probe-rs-cli download --format Bin --base-address 0x10000000 --chip rp2040 rp2040_updtfw_v1235_signed.bin` it is working.

imrank03, 2022-06-30:
It works, so I'm closing this issue.

closed 2022-06-30 by imrank03