rp-rs/rp-hal #379: I2C infinity loop

marius-meissner, 2022-07-08:
While debugging, I noticed that the application sometimes hangs forever in https://github.com/rp-rs/rp-hal/blob/0f5bc072cdb6d51bf27d6fd236fe2faf894124f9/rp2040-hal/src/i2c/controller.rs#L219 or https://github.com/rp-rs/rp-hal/blob/0f5bc072cdb6d51bf27d6fd236fe2faf894124f9/rp2040-hal/src/i2c/controller.rs#L227

The issue just seems to happen if using interrupts (RTIC application) and maybe is triggered if the second task is using SPI bus.

Unfortunately, I know too little about the internal workings to dig deeper here. But I'm happy to support if you tell me what information is needed.

jannic, 2022-07-10:
That's a tough one. It could be a bug in rp-hal, in the rest of your application, or perhaps even a hardware issue.

Can you extract some example which reproduces the issue, which you can share?

marius-meissner, 2022-07-11:
Seems the problem was related to either a hardware fault or high EMI environment.
After replacing the Pico board + reducing EMI sources, I was unable to reproduce the issue.

Thank you for your fast response.

closed 2022-07-11 by marius-meissner