rp-rs/rp-hal #382: Access the Pins by Index

TheFrozenDuck, 2022-07-10:
The documentation give a rather good overview of how to access pins using a pin's name. What I could not find though was how to access a pin by index. I would like to avoid matching like this:

```rust
match pin {
    0=>pins.gpio0.into_mode(),
    1=>pins.gpio1.into_mode(),
    2=>pins.gpio2.into_mode(),
    ...
```
Would that be possible to do?

jannic, 2022-07-10:
No, I don't think there is an easy way.
The current API makes sure that you can't get two handles to the same pin, at compile time. That would be difficult to achieve if you could somehow get the pin for a given pin number.
The closest thing I could imagine would be something like a function on `rp2040_hal::gpio::pin::bank0::Pins`:
```
pub fn to_dynpin_array(self) -> [DynPin; 30]
```
But currently no such thing exitsts, and I'm not sure how useful it would be.

TheFrozenDuck, 2022-07-10:
Thank you for the technological background. I agree with you that adding such thing is probably not utterly useful.

I only need it to do some multiplexing and that is problem not the most common use-case for this.

closed 2022-07-10 by TheFrozenDuck

koriwi, 2023-04-13:
Hope necromancing to prevent duplicate issues is ok.
In my usecase (freedeck-firmware) the user can set some pins which are stored in a config file on an sdcard. The only hard coded pins are the spi spins for the sdcard. the rest should be possible to be dynamic so people don't have to recompile everything just to change some pins. This is currently blocking me from porting this project to rust.

Maybe you have a solution for me or i gave you a good idea why it could be needed.

jannic, 2023-04-13:
IMHO adding comments to old tickets is ok. However they are not well visible, so it might easily happen that you don't get an answer.
About your question: I think your code, using pin_array, is fine. It has the big advantage of being safe, so you don't risk to accidentally use the same pin twice, concurrently.
If you really want to construct a pin type from its number, you'd need something like https://github.com/rp-rs/rp-hal/pull/482.  @9names, this PR has been in draft state for several months. Do you think it still needs work?

koriwi, 2023-04-13:
Hi Jannic. I hoped somebody would see it. If not i would have done something else :)
Yes. my code is safe which is what i ultimately want, so i will probably keep it that way. I saw the issue where this pr comes from. I don't really do interop stuff and i also do not like unsafe. Sooooo I'm probably not the guy to ask regardings this draft :P

jannic, 2023-04-13:
Ultimately, any safe function providing this functionality probably needs to contain a data structure similar to your `pin_array`. It might hold a compressed form (a bitmap of all available pins) in a single `u32` to reduce the memory footprint. But that would complicate the code, so it's a tradeoff.
There's room for a crate providing such tools, but personally I don't think it's a good fit for the HAL. It's a higher level concept.

koriwi, 2023-04-13:
I'm fine with that.
Maybe we could put a notice somewhere for noobies like me that come from Arduino or similar and are used to dynamically indexing pins. Was scratching my head a while before i came up with this :P