rp-rs/rp-hal #389: having difficulty accessing the monochrome OLED display of an Adafruit Macropad

mutantbob, 2022-07-17:
I have an Adafruit Macropad which I have been led to believe has an RP2040 microcontroller, and an SH1106G OLED display.

I have succeeded at using the neopixels via the `ws2812-pio` and `smart-leds-trait` crates.

Now I am trying to get something to display on the sh1106.

I have the following code:

```Rust

    let _ = pins
        .mosi
        .into_mode::<adafruit_macropad::hal::gpio::FunctionSpi>();
    let _ = pins
        .miso
        .into_mode::<adafruit_macropad::hal::gpio::FunctionSpi>();

    let spi1 = Spi::<_, _, 8>::new(pac.SPI1).init(
        &mut pac.RESETS,
        125_000_000u32.Hz(),
        1_000_000u32.Hz(),
        &MODE_0,
    );

    let mut disp: GraphicsMode<
        SpiInterface<
            Spi<_, _, 8_u8>,
            adafruit_macropad::hal::gpio::Pin<_, adafruit_macropad::hal::gpio::Output<Readable>>,
            adafruit_macropad::hal::gpio::Pin<_, adafruit_macropad::hal::gpio::Output<Readable>>,
        >,
    > = Builder::new()
        .connect_spi(spi1, pins.oled_dc.into_mode(), pins.oled_cs.into_mode())
        .into();

    disp.init().unwrap();
    disp.flush().unwrap();
    disp.clear();

    disp.draw_iter(
        (0..5)
            .map(move |y| {
                (0..10).map(move |x| {
                    embedded_graphics_core::Pixel(
                        embedded_graphics_core::prelude::Point::new(x, y),
                        if 0 == 1 & (x + y) {
                            embedded_graphics_core::pixelcolor::BinaryColor::On
                        } else {
                            embedded_graphics_core::pixelcolor::BinaryColor::Off
                        },
                    )
                })
            })
            .flatten(),
    );

    disp.flush();
```
But the display stays black.  

It would not surprise me if I have some of the frequency values wrong, or a template parameter that is incorrect.

I also have some text and line drawing stuff from the `embedded_graphics` crate, but it's a bit of a struggle, since the sh1106 seems to use the `embedded_graphics_core` crate, and those two crates (e-g, e-g-c) seem to have conflicting types named the same thing, so adapting between them requires shenanigans.

I was hoping that the adafruit_macropad crate would have some convenience functions that assemble the proper objects, but I have not yet found anything of the sort.

9names, 2022-07-18:
I have the display working in #340, might be useful to check what's different there.

sonnny, 2022-07-19:
Here is another link with macropad that is working for me --->  [https://github.com/dlkj/macropad-rust](https://github.com/dlkj/macropad-rust)

9names, 2022-07-19:
I was coming back to add it because I forgot to link to Daniel's stuff, thanks @sonnny!

mutantbob, 2022-07-21:
I have it working, and created the following macro.

```Rust
#[macro_export]
macro_rules! macropad_oled {
    ($pins:expr, $clocks:expr, $delay: expr, $pac:expr) => {{
        let _spi_sclk = $pins.sclk.into_mode::<$crate::FunctionSpi>();
        let _spi_sclk = $pins.mosi.into_mode::<$crate::FunctionSpi>();

        let spi1 = $crate::Spi::<_, _, 8>::new($pac.SPI1).init(
            &mut $pac.RESETS,
            $crate::Clock::freq(&$clocks.peripheral_clock),
            $crate::Extensions::Hz(16_000_000u32),
            // (16_000_000u32 as embedded_time::rate::Extensions).Hz(),
            &$crate::MODE_0,
        );

        let mut oled_reset = $pins.oled_reset.into_push_pull_output();

        let mut disp: $crate::GraphicsMode<_> = sh1106::Builder::new()
            .connect_spi(
                spi1,
                $pins.oled_dc.into_push_pull_output(),
                $pins.oled_cs.into_push_pull_output(),
            )
            .into();

        disp.reset(&mut oled_reset, &mut $delay).unwrap();

        disp.init().unwrap();
        disp
    }};
}
```

I created a lot of other macros to initialize the devices based on your showcase, and really should bundle them into a pull request for the `adafruit-macropad` crate.

closed 2022-07-21 by mutantbob

mutantbob, 2022-07-21:
pull request created: https://github.com/rp-rs/rp-hal/pull/392