rp-rs/rp-hal #393: Linker errors with intrinsics due to old builtins

9names, 2022-07-22:
This was originally raised in https://github.com/rp-rs/rp2040-project-template/issues/33
Hasn't been resolved yet, but posting here for greater visibility.

[yallxe](https://github.com/yallxe) commented
> error: linking with `flip-link` failed: exit code: 1
  |
  = note: "flip-link" "-flavor" "gnu" "D:\\Dev\\rp2040-project-template\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_project_template-1198b80032431cff.rp2040_project_template.9c75850d-cgu.0.rcgu.o" "--as-needed" "-L" "D:\\Dev\\rp2040-project-template\\target\\thumbv6m-none-eabi\\release\\deps" "-L" "D:\\Dev\\rp2040-project-template\\target\\release\\deps" "-L" "D:\\Dev\\rp2040-project-template\\target\\thumbv6m-none-eabi\\release\\build\\rp2040-project-template-ff201362c93239a8\\out" "-L" "D:\\Dev\\rp2040-project-template\\target\\thumbv6m-none-eabi\\release\\build\\cortex-m-edb3200537960378\\out" "-L" "D:\\Dev\\rp2040-project-template\\target\\thumbv6m-none-eabi\\release\\build\\cortex-m-rt-44339646bd76ccef\\out" "-L" "D:\\Dev\\rp2040-project-template\\target\\thumbv6m-none-eabi\\release\\build\\defmt-4ae0c0a354199f32\\out" "-L" "D:\\Dev\\rp2040-project-template\\target\\thumbv6m-none-eabi\\release\\build\\rp2040-pac-6d37e37f505878c4\\out" "-L" "C:\\Users\\yallxe\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-Bstatic" "C:\\Users\\yallxe\\AppData\\Local\\Temp\\rustcxHD7yj\\libcortex_m_rt-de909aa96899d9be.rlib" "--start-group" "C:\\Users\\yallxe\\AppData\\Local\\Temp\\rustcxHD7yj\\libcortex_m-f1aabfc11535f99f.rlib" "--end-group" "C:\\Users\\yallxe\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib\\libcompiler_builtins-8800918f309bcb9b.rlib" "-Bdynamic" "--eh-frame-hdr" "-znoexecstack" "-L" "C:\\Users\\yallxe\\.rustup\\toolchains\\nightly-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv6m-none-eabi\\lib" "-o" "D:\\Dev\\rp2040-project-template\\target\\thumbv6m-none-eabi\\release\\deps\\rp2040_project_template-1198b80032431cff" "--gc-sections" "-O1" "--nmagic" "-Tlink.x" "-Tdefmt.x"
  = note: rust-lld: error: duplicate symbol: __aeabi_uidivmod
          >>> defined at rp2040_project_template.9c75850d-cgu.0
          >>>            D:\Dev\rp2040-project-template\target\thumbv6m-none-eabi\release\deps\rp2040_project_template-1198b80032431cff.rp2040_project_template.9c75850d-cgu.0.rcgu.o:(.text+0x1)
          >>> defined at arm.rs:14 (/cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.66/src\arm.rs:14)
          >>>            compiler_builtins-8800918f309bcb9b.compiler_builtins.14216641-cgu.174.rcgu.o:(.text.__aeabi_uidivmod+0x1) in archive C:\Users\yallxe\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\thumbv6m-none-eabi\lib\libcompiler_builtins-8800918f309bcb9b.rlib
          
          rust-lld: error: duplicate symbol: __aeabi_idivmod
          >>> defined at rp2040_project_template.9c75850d-cgu.0
          >>>            D:\Dev\rp2040-project-template\target\thumbv6m-none-eabi\release\deps\rp2040_project_template-1198b80032431cff.rp2040_project_template.9c75850d-cgu.0.rcgu.o:(.text+0x35)
          >>> defined at arm.rs:82 (/cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.66/src\arm.rs:82)
          >>>            compiler_builtins-8800918f309bcb9b.compiler_builtins.14216641-cgu.174.rcgu.o:(.text.__aeabi_idivmod+0x1) in archive C:\Users\yallxe\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\thumbv6m-none-eabi\lib\libcompiler_builtins-8800918f309bcb9b.rlib
          
          flip-link: the native linker failed to link the program normally; please check your project configuration and linker scripts
          
The error mentions flip-link but the error is `rust-lld: error: duplicate symbol: __aeabi_uidivmod`
There's a fix for this in compiler_builtins-0.1.69 but as listed above it's using (compiler_builtins-0.1.66).

jannic, 2022-07-23:
Do you think we can do anything to fix this in rp-hal, or is the ticket more about properly documenting the required versions?

9names, 2022-07-24:
Neither of those reporting the bug specified their compiler versions, makes it hard to know if it was an old compiler issue.
It's both a documentation issue, plus a discovery one (so this issue will form done of the documentation).
Is there an easy way of knowing which compiler-builtins a particular rustc will use? We'll want both minimal stable plus nightly version.

9names, 2022-07-24:
I assume the best way to determine compiler-builtins version is to look at the Cargo.lock file in the rustc repo with each release tag?
In that case, rustc 1.60 is the minimum version if you want to have intrinsics feature enabled.
The version of builtin-intrinsics was bumped in this commit https://github.com/rust-lang/rust/commit/7b8f6ac5ab486d88afc6e80c8491ee5f6b5c4906 so any nightly 2022/02/11 or later should be fine. rustc release 1.60 was 2022/04/07.
So: best guess is this was getting built with an ancient nightly

9names, 2022-07-27:
I think this issue is now resolved, we'll link people here if they report this issue in the future.

closed 2022-07-27 by 9names