rp-rs/rp-hal #400: How do I get GPIO number of the pin that triggered an IO_IRQ_BANK0 Interrupt?

Altaflux, 2022-07-27:
Looking at the documentation for the RP2040 SDK and other examples it appears that one can retrieve that pin number that triggered the interrupt.

The function's type is as follows:
`typedef void(* gpio_irq_callback_t )(uint gpio, uint32_t event_mask)`

How could I obtain the same information from the interrupt?
The example included in this repository does not has that use case included:
https://github.com/rp-rs/rp-hal/blob/main/rp2040-hal/examples/gpio_irq_example.rs

9names, 2022-07-27:
That C function is a callback, it something that gets called *when* the event occurred.
The only way we have for testing this currently is to check the interrupt status for a GPIO.
We should update that example to demonstrate that. Example:
```rust
// Need to check if our Option<LedAndButtonPins> contains our pins
if let Some(gpios) = LED_AND_BUTTON {
// borrow led and button by *destructuring* the tuple
// these will be of type `&mut LedPin` and `&mut ButtonPin`, so we don't have
// to move them back into the static after we use them
let (led, button) = gpios;

if button.interrupt_status(EdgeLow) {
    // toggle can't fail, but the embedded-hal traits always allow for it
    // we can discard the return value by assigning it to an unnamed variable
    let _ = led.toggle();
    // Our interrupt doesn't clear itself.
    // Do that now so we don't immediately jump back to this interrupt handler.
    button.clear_interrupt(EdgeLow);
}
```

Altaflux, 2022-07-27:
thanks for the quick response, this does accomplish my needs!

closed 2022-07-27 by Altaflux