rp-rs/rp-hal #412: language item required, but not found: `eh_personality`

jareer12, 2022-08-03:
#33 Doesn't help, the fix mentioned there doesn't work.

**Error**
```shell
error: language item required, but not found: `eh_personality`
  |
  = note: this can occur when a binary crate with `#![no_std]` is compiled for a target where `eh_personality` is defined in the standard library
  = help: you may be able to compile for a target that doesn't need `eh_personality`, specify a target with `--target` or in `.cargo/config`

error: could not compile `RaspberryPiPicoRust` due to previous error
```

**Code**
```
//! # Pico GPIO In/Out Example
//!
//! Toggles the LED based on GPIO input.
//!
//! This will control an LED on GP25 based on a button hooked up to GP15. The
//! button should cause the line to be grounded, as the input pin is pulled high
//! internally by this example. When the button is pressed, the LED will turn
//! off.
//!
//! See the `Cargo.toml` file for Copyright and license details.

#![cfg_attr(not(test), no_std)]
#![no_main]

// The macro for our start-up function
use rp_pico::entry;

// GPIO traits
use embedded_hal::digital::v2::{InputPin, OutputPin};

// Ensure we halt the program on panic (if we don't mention this crate it won't
// be linked)
use panic_halt as _;

// A shorter alias for the Peripheral Access Crate, which provides low-level
// register access
use rp_pico::hal::pac;

// A shorter alias for the Hardware Abstraction Layer, which provides
// higher-level drivers.
use rp_pico::hal;

/// Entry point to our bare-metal application.
///
/// The `#[entry]` macro ensures the Cortex-M start-up code calls this function
/// as soon as all global variables are initialised.
///
/// The function configures the RP2040 peripherals, then just reads the button
/// and sets the LED appropriately.
#[entry]
fn main() -> ! {
    // Grab our singleton objects
    let mut pac = pac::Peripherals::take().unwrap();

    // Note - we don't do any clock set-up in this example. The RP2040 will run
    // at it's default clock speed.

    // The single-cycle I/O block controls our GPIO pins
    let sio = hal::Sio::new(pac.SIO);

    // Set the pins up according to their function on this particular board
    let pins = rp_pico::Pins::new(
        pac.IO_BANK0,
        pac.PADS_BANK0,
        sio.gpio_bank0,
        &mut pac.RESETS,
    );

    // Our LED output
    let mut led_pin = pins.led.into_push_pull_output();

    // Our button input
    let button_pin = pins.gpio15.into_pull_up_input();

    // Run forever, setting the LED according to the button
    loop {
        if button_pin.is_low().unwrap() {
            led_pin.set_high().unwrap();
        } else {
            led_pin.set_low().unwrap();
        }
    }
}

// End of file
```

**Cargo.toml**
```
[package]
name = "RaspberryPiPicoRust"
version = "0.1.0"
edition = "2018"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
cortex-m = "0.7.2"
rp2040-boot2 = { version = "0.2.0", optional = true }
rp2040-hal = { version = "0.5.0" }
cortex-m-rt = "0.7.1"
embedded-time = "0.12.0"
usb-device= "0.2.9"
usbd-serial = "0.1.1"
rp-pico = "0.4.0"
usbd-hid = "0.5.1"
futures = { version = "0.3", default-features = false, optional = true }
panic-halt= "0.2.0"
embedded-hal = "0.2.7"
```

9names, 2022-08-03:
> = help: you may be able to compile for a target that doesn't need `eh_personality`, specify a target with `--target` or in `.cargo/config`

Ah, the help line even says this now. Neat!

jareer12, 2022-08-03:
> 

```shell

   Compiling autocfg v1.1.0     
   Compiling proc-macro2 v1.0.43
   Compiling quote v1.0.21
   Compiling unicode-ident v1.0.2
   Compiling syn v1.0.99
   Compiling semver-parser v0.7.0
   Compiling cortex-m-rt v0.7.1
   Compiling nb v1.0.0
   Compiling cortex-m v0.7.5
error[E0463]: can't find crate for `core`
  |
  = note: the `thumbv6m-none-eabi` target may not be installed
  = help: consider downloading the target with `rustup target add thumbv6m-none-eabi`

error[E0463]: can't find crate for `compiler_builtins`

error[E0463]: can't find crate for `core`
   --> C:\Users\john\.cargo\registry\src\github.com-1ecc6299db9ec823\nb-1.0.0\src\lib.rs:189:5
    |
189 | use core::fmt;
    |     ^^^^ can't find crate
    |
    = note: the `thumbv6m-none-eabi` target may not be installed
    = help: consider downloading the target with `rustup target add thumbv6m-none-eabi`

error: cannot find attribute `derive` in this scope
   --> C:\Users\john\.cargo\registry\src\github.com-1ecc6299db9ec823\nb-1.0.0\src\lib.rs:198:3
    |
198 | #[derive(Clone, Copy, PartialEq, Eq, PartialOrd, Ord, Hash)]
    |   ^^^^^^

error[E0463]: can't find crate for `core`
   --> C:\Users\john\.cargo\registry\src\github.com-1ecc6299db9ec823\nb-1.0.0\src\lib.rs:192:27
    |
192 | pub type Result<T, E> = ::core::result::Result<T, Error<E>>;
    |                           ^^^^ can't find crate
    |
    = note: the `thumbv6m-none-eabi` target may not be installed
    = help: consider downloading the target with `rustup target add thumbv6m-none-eabi`

error[E0405]: cannot find trait `FnOnce` in this scope
   --> C:\Users\john\.cargo\registry\src\github.com-1ecc6299db9ec823\nb-1.0.0\src\lib.rs:223:12
    |
223 |         F: FnOnce(E) -> T,
    |            ^^^^^^ not found in this scope

error[E0405]: cannot find trait `From` in this scope
   --> C:\Users\john\.cargo\registry\src\github.com-1ecc6299db9ec823\nb-1.0.0\src\lib.rs:232:9
    |
232 | impl<E> From<E> for Error<E> {
    |         ^^^^ not found in this scope

Some errors have detailed explanations: E0405, E0463.
For more information about an error, try `rustc --explain E0405`.
error: could not compile `nb` due to 7 previous errors
warning: build failed, waiting for other jobs to finish...
```

9names, 2022-08-03:
Did you try what the help text said for this error?
We also have some steps in the "getting started" section of the readme that you should do before going any further.
https://github.com/rp-rs/rp-hal#getting-started

waalge, 2022-08-07:
I have 
```
$ rustup target add thumbv6m-none-eabi
info: component 'rust-std' for target 'thumbv6m-none-eabi' is up to date
```
but I also get the above.

9names, 2022-08-07:
interesting. are you able to build https://github.com/rp-rs/rp2040-project-template/
if you could provide a link to your current project I could also try testing it out locally.

waalge, 2022-08-07:
@9names That's very kind, thanks! (Also very speedy) 

I'm running nixos, and I think there's something funky in my setup. 
If it persists I'll try to share something diagnosable... and if it doesn't I'll share what steps resolved it

waalge, 2022-08-08:
I will try to say more helpful things when I do this again from scratch.
I did eventually flash the device but only after changing and unchanging and installing and uninstalling too many things... I'm not sure what actually solved my issues. And it was issues. 

Couple of things 
- `~/.cargo/bin` was not on my `$PATH`. Not sure how it fell off and took me some time to find this. 
- I switched to using https://github.com/rp-rs/rp2040-project-template
- I think I've previously used `probe-run` as the runner. I couldn't get this to work. The pico was only being detected if in boot mode as mass storage device. probe-run would say no device found and that this is usually a cable or permission problem. I havent yet understood quite how pico flashing is supposed to work. Mounting it manually as mass storage device, and then using the runner elf2uf2-rs.

9names, 2022-08-08:
I wonder how anything was working if ~/.cargo/bin wasn't on your path, that's quite strange.

I can probably help understand the last point: probe-run is designed to work with a dedicated hardware debug probe (a device that talks to pico over the debug pins on the opposite end from the USB connector). If you don't have one of those, you can't use probe-run to flash a new program into the Pico's non-volatile flash memory.
elf2uf2-rs is the right tool for running code if you are putting the Pico in bootloader mode by holding the boot button when connecting it via USB.

closed 2023-10-06 by jub0t