rp-rs/rp-hal #446: Dynamic GPIO pin with static mode.

agausmann, 2022-09-04:
# Motivation 

In AVR HALs, it is possible to [downgrade a pin handle](https://rahix.github.io/avr-hal/avr_hal_generic/port/struct.Pin.html#downgrading) so that the pin number is erased from the type, but the input/output typestate is still preserved.

This makes it possible to create an array of output pins like `[Pin<Output, Dynamic>; 12]`, and it is guaranteed that the pins are in output mode, so that the `set_low` and `set_high` methods are infallible.

rp2040-hal has the `DynPin` API but it isn't quite the same. It is dynamic over _both_ the mode and the pin number, and there doesn't seem to be a way to make only one of those parameters dynamic. This comes with a couple disadvantages:

- The pin mode isn't enforced at compile time. If I wanted to create an array of output pins like the previous example, the type would be `[DynPin; 12]` and it would not be a compile error to construct it with an expression like `[pins.gpio2.into(), pins.gpio5.into(), ...]`, which is  forgetting to set the pin states.

- The `is_low`/`is_high`/`set_low`/`set_high` methods on `DynPin` are fallible and could return `Err` if the pin is in the wrong mode. This is not a big deal, as the error variants of `InputPin` and `OutputPin` APIs would have to be handled/ignored anyway even if they were infallible.

9names, 2022-09-04:
We derived our GPIO impl from atsamd, I think we should get those folks involved if we're adding a new type of Pin to the implementation, at the very least to get their opinion on naming and such.
https://github.com/atsamd-rs/atsamd/blob/master/hal/src/gpio/mod.rs

ithinuel, 2022-09-04:
From the feedback and questions I've seen here and on our matrix channel, it seems like in addition to the fully typestate'd pin and the completely dyn-typed pins, we'd need:

- 1 type that erases the identity of the pin while validating its state so that, for examples, arrays of input pins can be built.
- 1 type that erases only the state of the pin while keeping its identity, so that a module can change its direction at runtime.

closed 2023-05-29 by ithinuel