rp-rs/rp-hal #456: pico_enable_stdio_usb

mark2b, 2022-09-20:
Using pico_sdk for cpp, there is feature to enable USB stdio in the MakeLists.txt
**pico_enable_stdio_usb(${NAME} 1)**
It allows use picotool not in BOOTSEL mode
Is rp-hal has similar feature?

ithinuel, 2022-09-20:
Hi, 

If my understanding is correct, this CMake function enables the use of a usb CDC as the std in/out (`printf`).

When this is enabled, a side effect [by default](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/include/pico/stdio_usb.h#L80-L83) is also to add a [`class vendor specific` interface](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/stdio_usb_descriptors.c#L97-L99) (implemented [here](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_stdio_usb/reset_interface.c)) that allows a program on the host to trigger a reset to usb boot.

Picotools [searches for this interface](https://github.com/raspberrypi/picotool/blob/master/main.cpp#L2025-L2064) and ignores the others, so you don't need the first ones to be CDC (it could even be the only usb interface present on the device).

I don't think such a feature belongs to the rp-hal _per se_ (as it would require to setup the usb peripheral and to poll the usb stack in the application anyways) but it could definitely be implemented as a separate crate implementing the vendor specific [UsbClass](https://docs.rs/usb-device/latest/usb_device/class/trait.UsbClass.html) to behave like the sdk version does. I don't know if anyone ever tried to do that.

romainreignier, 2024-09-13:
I was looking for the same feature and I have seen that @ithinuel has worked on it: https://github.com/ithinuel/usbd-picotool-reset

Tested with `picotool info -f -a`, the device effectively reboots into BOOTSEL mode but might take too much time because picotool does not detect it soon enough:

```
$ picotool info -f -a
Tracking device serial number TEST for reboot
The device was asked to reboot into BOOTSEL mode so the command can be executed.

Despite the reboot attempt, no accessible RP2040/RP2350 devices in BOOTSEL mode were found. It is possible the device is not responding, and will have to be
manually entered into BOOTSEL mode.
```

But the device is indeed in BOOTSEL mode:

```
$ picotool info -a
Program Information
 none

Fixed Pin Information
 none

Build Information
 none

Device Information
 type:        RP2040
 revision:    B1
 flash size:  2048K
 flash id:    0xE66118604B286D24
```

And we can reboot it back to application code with:
```
picotool boot -a
```