rp-rs/rp-hal #458: RTIC scheduled software tasks consume TIMER, preventing use of Timer/Alarm hardware interrupts

xabra, 2022-09-24:
When using RTIC, if one wants to be able to schedule a software task (eg `taskname::spawn_after(duration)`) it is necessary to bind the TIMER peripheral to the monotonic:  ` let mono = Rp2040Monotonic::new(c.device.TIMER);`.  This makes the Timer peripheral and its four  hardware Alarm interrupts unavailable.

It would be useful to have Timer/Alarm hardware interrupts _and_ scheduled software tasks available simultaneously.  For example, to set key timing cycles using low overhead HW interrupts, and schedule SW tasks relative to them.

Perhaps deriving the monotonic from SYSTICK rather then TIMER would work.

ithinuel, 2022-09-24:
Thanks for reporting this.

the implementation of monotonic you are refering to lives in an other repository .

Would you mind giving a try at : https://github.com/ithinuel/rp-hal/tree/impl-rtic-monotonic
When the feature `rtic-monotonic` is enabled, this adds a new type under [`rp2040_hal::timer::monotonic`](https://github.com/ithinuel/rp-hal/blob/impl-rtic-monotonic/rp2040-hal/src/timer.rs#L386-L430) that supports rtic's monotonic as well as taking any of the 4 alarm types while still giving access to the under lying timer to take other alarms if/when needed.

If you are already using the latest hal you should be able to add the following snippet to your `Cargo.toml` (you may face breakage on some components, like pio, that have evolved since last release):

```toml
[patch.crates-io]
rp2040-hal = { git = "https://github.com/ithinuel/rp-hal", branch = "impl-rtic-monotonic" }
rp-pico = { git = "https://github.com/ithinuel/rp-hal", branch = "impl-rtic-monotonic" }
# or-another-bsp = { git = "https://github.com/ithinuel/rp-hal", branch = "impl-rtic-monotonic" }
```

If this works for you I'll make a PR out of it.

jannic, 2022-09-24:
[rp2040_hal::timer::monotonic](https://github.com/ithinuel/rp-hal/blob/impl-rtic-monotonic/rp2040-hal/src/timer.rs#L386-L430) contains:
```
        unsafe fn reset(&mut self) {
            todo!()
        }
```
Doesn't this immediately panic? `reset` is [documented](https://docs.rs/rtic-monotonic/latest/rtic_monotonic/trait.Monotonic.html#tymethod.reset) as being called by the RTIC runtime during initialization.

ithinuel, 2022-09-24:
Yes, sorry I noticed it later, I just pushed a fix for that.
The implementation is likely to require some polish. I tried modifying the `pico_rtic` example but that's litteraly my first rtic experience so I don't know if I'm doing something wrong or if my impl is broken (both are as likely so far).

xabra, 2022-09-25:
> Thanks for reporting this.
>
> the implementation of monotonic you are refering to lives in an other repository .
>
> Would you mind giving a try at : [...]

I'm sorry but I cant get it to compile.  Probably my ignorance.
I just copied your example code and modified my .toml file per the above.  But I get:
**error[E0432]: unresolved import `rp_pico::hal::timer::monotonic`
  --> src\main.rs:16:21
   |
16 |             timer::{monotonic::Monotonic, Alarm0},
   |                     ^^^^^^^^^ could not find `monotonic` in `timer`**

Also tried to turn on the feature switch like this:
[patch.crates-io]
rp2040-hal = { git = "https://github.com/ithinuel/rp-hal", branch = "impl-rtic-monotonic" , feature = "rp2040-hal/rtic-monotonic" }
rp-pico = { git = "https://github.com/ithinuel/rp-hal", branch = "impl-rtic-monotonic" }

ithinuel, 2022-09-25:
I'm not sure `feature = "name"` works, you can use `features = [ "rtic-monotonic" ]`

Alternatively you can force it from the command line with `--features rtic-monotonic`.

xabra, 2022-09-25:
Thank you!  Your example code compiles and runs perfectly.  
I will roll this new functionality into my project code so it uses both HW tasks and scheduled SW tasks 
I will let you know what happens.  But I can't imagine why it wouldn't work

xabra, 2022-09-26:
Hi @ithinuel,
In your example code you spawned a scheduled software task using monotonic bound to Alarm0.
I made test code that still uses Alarm0 for the monotonic (no SW task yet), but also has a hardware task bound to Alarm1/TIMER_IRQ_1 here:
https://gist.github.com/xabra/47b8ab45f50bdc4cf28cc9c0b79849d6
It doesn't compile because:

```
error[E0382]: use of moved value: `timer`
  --> src\main.rs:80:90
   |
67 |         let mut timer = hal::Timer::new(c.device.TIMER, &mut resets);
   |             --------- move occurs because `timer` has type `Timer`, which does not implement the `Copy` trait
...
80 |         (Shared { timer, alarm1, measure_pin}, Local {}, init::Monotonics(Monotonic::new(timer, alarm0)))
   |                   ----- value moved here                                                 ^^^^^ value used here after move
```

jannic, 2022-09-26:
You don't need the `timer` in `Shared`. If you remove that, it compiles:
```
--- a/boards/rp-pico/examples/pico_monotonic_test.rs
+++ b/boards/rp-pico/examples/pico_monotonic_test.rs
@@ -24,7 +24,6 @@ mod app {
 
     #[shared]
     struct Shared {
-        timer: hal::Timer,
         alarm1: hal::timer::Alarm1,     // HW IRQ 
         //alarm2: hal::timer::Alarm2,   // Unused for now
         //alarm3: hal::timer::Alarm3,   // Unused for now
@@ -76,7 +75,7 @@ mod app {
         let _ = alarm1.schedule(SCAN_TIME_US);
         alarm1.enable_interrupt();
 
-        (Shared { timer, alarm1, measure_pin}, Local {}, init::Monotonics(Monotonic::new(timer, alarm0)))
+        (Shared { alarm1, measure_pin}, Local {}, init::Monotonics(Monotonic::new(timer, alarm0)))
     }
 
     // -- TASK: High speed measurement & processing cycle start
```

xabra, 2022-09-26:
You are right.  It runs.  Very Cool.

xabra, 2022-09-27:
@ithinuel @jannic @9names:
I tested ithinuel's implementation of RTIC Monotonic with code that uses both hardware tasks and scheduled software tasks simultaneously.  The former triggers the latter to make a simple PWM.  

It works perfectly.  Well done.  Kudos to you.  

My test code is here:
https://gist.github.com/xabra/4061867af0af536f2bd30be1c848620c

I put a lot of comments in the code in case someone wants to use it as an example at some point.  They would have saved me a lot of confusion about systick, monotonic, etc.

As a novice, can you explain to me the process by which such improvements get merged onto the main branch of rp-hal and ultimately into crates.io?

ithinuel, 2022-09-27:
@xabra Thank you so much for your feedback!

Your test code would make a perfect example :) 

The general process is:
- for the repository you want to contribute to
- clone your fork on your workstation and implement the changes you want
- push them to a branch on your fork
- Github should invite you to create a PR from your new branch to rp-rs/main
- From there we will review (and may request some extra changes) and merge

The publication to crates.io is then a matter of time. There's no set rules (as far as I know) about when we publish new release.
I'd say this is when enough changes have landed on main and/or when people ask nicely 🙂

ithinuel, 2022-10-06:
This has now been merged in https://github.com/rp-rs/rp-hal/pull/459

@xabra are you ok with this issue being closed?

xabra, 2022-10-07:
@ithinuel - Yes.  OK to close. 
Thank you.

closed 2022-10-07 by ithinuel