rp-rs/rp-hal #466: PIO: No documented way to use same program for more SMs

Riphiphip, 2022-10-11:
I have a PIO program that I want to run on all 4 SMs on PIO0. When using PIOBuilder with an instance of InstalledProgram, the PIOBuilder takes ownership over the InstalledProgram, preventing it from being reused for other SMs. Since my program uses 10 instruction I don't have enough instruction memory to install four copies of it. Am I missing something or is this a mistake?

This snippet causes my program to panic at the install related to sm3. Commenting out everything that has to do with sm3 allows the program to run nicely.

```rust
  let (mut pio0, sm0, sm1, sm2, sm3) = pac.PIO0.split(&mut pac.RESETS);

  let pulse_width_program = {
      use pio::JmpCondition::{PinHigh, XDecNonZero};
      use pio::SetDestination::{PINDIRS, PINS};
      use pio::WaitSource::PIN;
      use pio::{
          MovDestination::{ISR, X as XDst},
          MovOperation::{Invert, None},
          MovSource::{NULL, X as XSrc},
      };

      let mut a = pio::Assembler::new();
      let mut wrap_target = a.label();
      let mut wrap_source = a.label();

      let mut x_not_zero = a.label();

      let mut wait_loop_top = a.label();
      let mut wait_loop_bottom = a.label();

      a.set(PINDIRS, 0b10); // Set pin 0 to input and pin 1 to output
      a.bind(&mut wrap_target);
      // Initialize registers
      a.mov(XDst, Invert, NULL); // Fill X with ones
      a.wait(1, PIN, 0, false); // Wait for pin 0 to go high

      // Count number of "peripheral cycles" that pin is high
      a.bind(&mut wait_loop_top);
      a.set(PINS, 0b10);
      a.jmp(XDecNonZero, &mut x_not_zero); // Decrement X
      a.bind(&mut x_not_zero);
      a.jmp(PinHigh, &mut wait_loop_top); // If pin is still high, keep waiting
      a.bind(&mut wait_loop_bottom);

      a.set(PINS, 0b00);
      // Move counter to CPU
      a.mov(ISR, None, XSrc); // Move content of X register to ISR
      a.push(false, false); // Push to CPU

      a.bind(&mut wrap_source); // Wrap

      a.assemble_with_wrap(wrap_source, wrap_target)
};

let (sm0_init, mut rc_chan_0, _) = {
        let installed = pio0.install(&pulse_width_program).unwrap();
        PIOBuilder::from_program(installed)
            .in_pin_base(rc_chan_0_pin_id)
            .jmp_pin(rc_chan_0_pin_id)
            .clock_divisor(2.0)
            .build(sm0)
    };

    let (sm1_init, mut rc_chan_1, _) = {
        let installed = pio0
            .install(&pulse_width_program)
            .expect("Could not install to sm1");
        PIOBuilder::from_program(installed)
            .in_pin_base(rc_chan_1_pin_id)
            .jmp_pin(rc_chan_1_pin_id)
            .clock_divisor(2.0)
            .build(sm1)
    };

    let (sm2_init, mut rc_chan_2, _) = {
        let installed = pio0
            .install(&pulse_width_program)
            .expect("Could not install to sm2");
        PIOBuilder::from_program(installed)
            .in_pin_base(rc_chan_2_pin_id)
            .jmp_pin(rc_chan_2_pin_id)
            .clock_divisor(2.0)
            .build(sm2)
    };

    let (sm3_init, mut rc_chan_3, _) = {
        let installed = pio0
            .install(&pulse_width_program)
            .expect("Could not install to sm3");
        PIOBuilder::from_program(installed)
            .in_pin_base(rc_chan_3_pin_id)
            .jmp_pin(rc_chan_3_pin_id)
            .clock_divisor(2.0)
            .build(sm3)
    };
```

Riphiphip, 2022-10-11:
Nvm, I figured it out. Calling InstalledProgram.share() creates a new instance of InstalledProgram with the same parameters, although I would recommend updating the [pio_synchronized](https://github.com/rp-rs/rp-hal/blob/39190c357f95e84947597c9f4d47193c941e965c/rp2040-hal/examples/pio_synchronized.rs#L74) sample to better explain this. It even has a comment expressing confusion about this.

jannic, 2022-10-11:
Hi @Riphiphip, thanks for your report!
Do you think https://github.com/jannic/rp-hal/commit/e9923cbcd2da55c69e1ea563085bb1ad8e37d5aa would be a good fix for that example?

Riphiphip, 2022-10-11:
Yes, that looks good!

I have however started to think  that it might be a good idea to create some sort of builder-pattern for SMs that share a program, as I suspect that it is a quite common use-case.

Riphiphip, 2022-10-11:
Accidentaly closed this. Reopening until a potential merge (or until someone else feels like closing it)

jannic, 2022-10-12:
> I have however started to think that it might be a good idea to create some sort of builder-pattern for SMs that share a program, as I suspect that it is a quite common use-case.

Yes, `InstalledProgram::share()` is a little bit ugly, it would be nice if it was possibly to share code without using `unsafe`.
To allow for uninstalling shared PIO programs safely, some additional accounting would be necessary, so this wouldn't be exactly zero-cost. But it should only take a few bytes of RAM.

PR #467 (2022-10-25): Use InstalledProgram::share() in examples/pio_synchronized.rs
Fixes #466
closed 2022-10-25 by jannic