rp-rs/rp-hal #475: How to handle multiple gpio interrupts

lucazulian, 2022-10-19:
I need to handle multiple gpio interrupts (the case is two encoders with two signals for each encoder). Following the example I created a type:

```rust
pub type ENCODER_DX_A = gpio::Pin<gpio::bank0::Gpio8, gpio::PullUpInput>;
pub type ENCODER_DX_B = gpio::Pin<gpio::bank0::Gpio9, gpio::PullUpInput>;
pub type ENCODER_SX_A = gpio::Pin<gpio::bank0::Gpio6, gpio::PullUpInput>;
pub type ENCODER_SX_B = gpio::Pin<gpio::bank0::Gpio7, gpio::PullUpInput>;

pub type ENCODERS = (ENCODER_DX_A, ENCODER_DX_B, ENCODER_SX_A, ENCODER_SX_B);
```
a static variable:

```rust
 static GLOBAL_ENCODERS: Mutex<RefCell<Option<ENCODERS>>> = Mutex::new(RefCell::new(None));
```

and their configuration:

```rust
  let encoder_dx_a: ENCODER_DX_A = pins.gpio8.into_mode();
  encoder_dx_a.set_interrupt_enabled(EdgeHigh, true);

  let encoder_dx_b: ENCODER_DX_B = pins.gpio9.into_mode();
  encoder_dx_b.set_interrupt_enabled(EdgeHigh, true);

  let encoder_sx_a: ENCODER_SX_A = pins.gpio6.into_mode();
  encoder_sx_a.set_interrupt_enabled(EdgeHigh, true);

  let encoder_sx_b: ENCODER_SX_B = pins.gpio7.into_mode();
  encoder_sx_b.set_interrupt_enabled(EdgeHigh, true);

  critical_section::with(|cs| {
      GLOBAL_ENCODERS.borrow(cs).replace(Some((
          encoder_dx_a,
          encoder_dx_b,
          encoder_sx_a,
          encoder_sx_b,
      )));
  });
```
enabled the interrupts:

```rust
    unsafe {
        pac::NVIC::unmask(pac::Interrupt::IO_IRQ_BANK0);
    }
```

and this is the handler:

```rust
#[allow(non_snake_case)]
#[interrupt]
fn IO_IRQ_BANK0() {
    static mut ENCODERS_DX_SX: Option<ENCODERS> = None;

    if ENCODERS_DX_SX.is_none() {
        critical_section::with(|cs| {
            *ENCODERS_DX_SX = GLOBAL_ENCODERS.borrow(cs).take();
        });
    }

    critical_section::with(|cs| {
        FOO.replace(cs, 12);
    });

    if let Some(gpios) = ENCODERS_DX_SX {
        let (encoder_dx_a, encoder_dx_b, encoder_sx_a, encoder_sx_b) = gpios;

        if encoder_dx_a.interrupt_status(EdgeHigh) {
            encoder_dx_a.clear_interrupt(EdgeHigh);
        }

        if encoder_dx_b.interrupt_status(EdgeHigh) {
            encoder_dx_b.clear_interrupt(EdgeHigh);
        }

        if encoder_sx_a.interrupt_status(EdgeHigh) {
            encoder_sx_a.clear_interrupt(EdgeHigh);
        }

        if encoder_sx_b.interrupt_status(EdgeHigh) {
            encoder_sx_b.clear_interrupt(EdgeHigh);
        }
    }
}
```
The loop contains some logic
```rust
loop {
        my_logic.run()
        // cortex_m::asm::wfi();
    }
```
The problem here is that it seems that the interrupts work only if I uncomment the line `cortex_m::asm::wfi();`. This is a problem since my logic should not be related to the interrupt events. Do you know how to make interrupts work without waiting in the loop for these?

jannic, 2022-10-19:
What do you mean by "they only work"? If I'm not missing something, your interrupt handler doesn't do anything besides clearing the interrupt flags. So what should happen when an interrupt fires? What happens instead if you don't call wfi?

lucazulian, 2022-10-19:
Sorry, in the `IO_IRQ_BANK0` function there is this:
```rust
    critical_section::with(|cs| {
        FOO.replace(cs, 12);
    });
```

and in my `my_logic.run()` I print its value. But without `cortex_m::asm::wfi()` I don't see FOO = 12

jannic, 2022-10-19:
Ah yes, sorry I missed that line, probably because I expected some action based on the actual values of the GPIO lines.

That's strange indeed. AFAIK that `wfi` should not be needed. So what I'd expect to see is:

1) With `wfi`: `my_logic.run()` may be called once before the interrupt fires, and then exactly once for each interrupt. So you should see the initial value of FOO zero or one times, and then the value 12.
2) Witout `wfi`: `my_logic.run()` may be called multiple times before and after the interrupt fires. So you may see the initial value of FOO many times, but eventually, once the interrupt got triggered, it should change to 12.

But I agree, the `wfi` instruction should not be needed for the interrupt to work.
Is there something interesting in `my_logic.run()` which could break things if it's called too often before the interrupt gets triggered the first time?

lucazulian, 2022-10-19:
This is my real loop
```rust
    loop {
        critical_section::with(|cs| {
            let foo = FOO.borrow(cs);
            let val = foo.borrow_mut();

            my_logic.error(*val);
        });

        my_logic.tick(timer.get_counter());
    }
```

`my_logic.error` just print in the USB serial port

jannic, 2022-10-19:
Just a guess: Without the `wfi`, the loop will run very often. So the printing to USB will for sure happen faster than the data can be actually transferred. What happens when the transfer buffer is full? If you just block on the `write` call, that won't work, as the buffer would never be emptied unless `poll` gets called repeatedly.

lucazulian, 2022-10-19:
What I expect in the loop from the example is to see in the USB the number 12, but I move the encoder, and I see 0. With wfi I would see 12.