rp-rs/rp-hal #478: BOOTSEL Button Status

brxken128, 2022-10-20:
This may sound odd, but is reading the status of `BOOTSEL` planned at any point?

[This](https://github.com/earlephilhower/arduino-pico/blob/0afd6ae54f88b8d62fa0ff2fa371c28ea7aecf0c/cores/rp2040/Bootsel.cpp) is the only information I can find about getting the status, and I have no clue how/if this could be translated to Rust with the current tooling available.

I think it'd be pretty helpful, especially as a quick/easy way to run an action, but I understand there are some downsides (disabling both interrupts and flash access). I'm not even sure this is possible within Rust.

Any pointers, thoughts, or guidance would be very greatly appreciated - thank you.

9names, 2022-10-20:
If you're willing to run entirely from ram and never access flash this is possible already, see https://github.com/9names/rp2040_rust_playground/tree/main/bootsel
As for disabling flash, interrupts and the 2nd core, this is essentially the same problem as https://github.com/rp-rs/rp-hal/issues/257, and could be solved in the same way

brxken128, 2022-10-20:
Thank you so much! I had just started trying to figure it out and was writing the code to access the `Cs` pin as you replied haha.

This is extremely helpful - thank you for the pointers and apologies for not hunting hard enough myself. Searching online just brought me to articles reminding users to hold BOOTSEL to flash/upload new programs.

closed 2022-10-20 by brxken128