rp-rs/rp-hal #481: Provide a way to construct a DynPin (for C interop)

9names, 2022-10-23:
cr1901 asked on #rust_embedded if there was a way to construct a pin for use with C code.
As in: existing C project. new code written in Rust, assuming it's called through an `extern "C" fn`

We're almost able to construct a DynPin from scratch now, but the DynRegisters struct is private.

Here's an example of the *almost working* syntax we currently have:
```rust
let pin = DynPinId {
    group: Bank0,
    num: 1,
};
let mode = DynPinMode::Output(PushPull);
// This doesn't work as DynRegisters is private
let regs = DynRegisters::new(pin);
let dynpin1 = DynPin { regs, mode };
dynpin1.toggle();
```

It's a bit verbose.
If we were to provide such a facility (and I think we should) how should we enable this?
Should we add an (unsafe) function that allows you to conjure a DynPin based solely on pin number and mode, along with documentation on how it will explode if you get these wrong?
Would appreciate any thoughts on whether we should do this, and how.

cr1901, 2022-10-23:
For more context: the idea is that the C code provides a header that defines the pin layout. I will parse this on the Rust side to create structs for appropriately initializing the used hardware on the Rust side (and provide a context pointer back to the C side after init is done).

Since there's no real way to enforce this on the C side, it's up to the C side to not call hw init functions that Rust is using.

jannic, 2022-10-23:
Without commenting on the implementation details, as I didn't have time to read the code yet: l think this is a valid use case and should be supported somehow. If it's only possible with some (unsafe) function which crates the DynPin out of thin air, so be it.

Are there alternatives? What about some container storing the real DynPins on the rust side, and handling the pin id as some kind of handle? That way C code could refer to the pin by its id, but could still only access pins which were previously registered in that structure. But that's probably just overcomplicating the issue.