rp-rs/rp-hal #501: Async / executer

mschnell1, 2022-11-20:
As the dual core paradigm is one of the key features of the Pico / 2040, Rust support would need to allow for async fn to be executed in parallel without introducing an operation system in the project but by simply distributing the tasks to the two cores with the "executer".
Is there already anything planned with such a goal ?
Thanks a lot for listening,
-Michael

jannic, 2022-11-20:
As we don't have async support in rp-hal yet, the question of how the schedule async tasks to both cores isn't the most pressing one.

If you are interested in async rust on the rp2040, it's probably worth to have a look at https://github.com/embassy-rs/embassy. However, I have no idea if they have multi-core support.

mschnell1, 2022-11-20:
In fact, I am interested in concurrent execution on the cores without the need of an OS. 
In fact it does not seem too intimidating to do an async executer that **only** supports two concurrent tasks by means of the cores. I'll have a look, once I actually am able to write fluently Rust code :)

ithinuel, 2022-11-20:
For what it's worth, I am running a nostd_async executor on each core on wilskeeb.
https://github.com/ithinuel/wilskeeb/blob/develop/firmware/src/main.rs#L544-L562

Tasks don't get passed from one core to the other because that'd involve some extra requirement that I didn't want to spend time on.

mschnell1, 2022-11-20:
Am I wrong assuming, that by "async fn" and "future" in principle any type of real or virtual parallelism can be handled, and it depends on the executor (using an OS, if present) which kind is in charge (e.g. single core scheduler, SMP, or - e.g. with 2040 (without OS and low complexity executor) , dual core (limited to two tasks) ? 
-Michael

9names, 2022-11-20:
In this context: async gets you concurrency, not parallelism.