rp-rs/rp-hal #509: the name `_critical_section_1_0_acquire` is defined multiple times

9names, 2022-11-30:
We had to update our version of critical_section, since critical_section 0.2.0 will silently break when mixed with critical_section 1.0. 
If you see the above error message, you need to update your dependency to rp2040-hal v0.6.1 (or downgrade to an older version of the HAL)
Sorry for any inconvenience!

Full error for anyone who needs more context:
```system
 Compiling rp2040-hal v0.6.0 (/home/nine/vc/rust/rp2040/rphal-update/rp2040-hal)
error[E0428]: the name `_critical_section_1_0_acquire` is defined multiple times
 --> rp2040-hal/src/critical_section_impl.rs:5:1
  |
5 | critical_section_0_2::custom_impl!(RpSpinlockCs);
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `_critical_section_1_0_acquire` redefined here
6 | #[cfg(feature = "critical-section-impl")]
7 | critical_section::set_impl!(RpSpinlockCs);
  | ----------------------------------------- previous definition of the value `_critical_section_1_0_acquire` here
  |
  = note: `_critical_section_1_0_acquire` must be defined only once in the value namespace of this module
  = note: this error originates in the macro `critical_section_0_2::custom_impl` (in Nightly builds, run with -Z macro-backtrace for more info)
```

Please comment or emote on this issue if you've hit this problem so we have an idea of how many people are affected.
Also: feel free to ask if you don't know how to resolve this.

jannic, 2023-05-14:
I think this change old enough that we can close the ticket now. We can leave it pinned so it'll still show up on top of https://github.com/rp-rs/rp-hal/issues. @9names, do you agree?

9names, 2023-05-14:
Sounds good to me

closed 2023-05-14 by jannic