rp-rs/rp-hal #520: SPI blocking write API does not match CS line behavior of spi_write_blocking in the pico-sdk

knutaf, 2022-12-12:
`spi_write_blocking` in the pico-sdk transfers the output bytes into the SPI DR register, doing a tight loop until all bytes have been written. This part is the same as the implementation in this crate. However, as a next step, it does a tight loop waiting until the SR register does not have `SPI_SSPSR_BSY_BITS`. The implementation in this crate doesn't--it just returns. Should `embedded_hal::blocking::spi::Write::write` also wait for not busy, to match behavior?

I've spent the last day debugging this issue when interfacing a Raspberry Pi Pico with an ILI9488 LCD driver, where the CS line is transitioning between low and high in between random bytes I'm sending, rather than in between each byte. I think it's confusing the hardware. By inserting `while spi.is_busy() {}` after each `write` call, I was able to fix my issues and make it work.

I was surprised that the "blocking" API didn't take care of this automatically. If there are reasons not to do it automatically, a parameter or flag somewhere, or at least added documentation might help others who need this deterministic CS line behavior.

jannic, 2022-12-12:
You are right that this is confusing when coming from the C SDK. However this function is coming directly from embedded_hal, it's not rp2040 specific code.

The documentation of embedded-hal 0.2.7 doesn't make it explicit, but for embedded-hal 1.0.0, the documentation says:
"To improve performance, [SpiBus](https://docs.rs/embedded-hal/1.0.0-alpha.9/embedded_hal/spi/trait.SpiBus.html) implementations are allowed to return before the operation is finished, i.e. when the bus is still not idle."
https://docs.rs/embedded-hal/1.0.0-alpha.9/embedded_hal/spi/index.html#flushing

jannic, 2022-12-12:
I still wonder if there is something else going on:
While `write` doesn't wait for the busy flag to clear, it does block on the reading side before returning: https://docs.rs/embedded-hal/latest/src/embedded_hal/blocking/spi.rs.html#72

How is it possible for the read to finish before the write is done?

The only explanation I can thin of is that you already had some data in the read buffer when calling `write`.

jannic, 2022-12-12:
(Others around here are more knowledgeable than me - please correct me if I'm wrong.)

The SPI interface sends a single bit and receives a single bit per clock cycle. So the amount of data sent and received is always equal. On a unidirectional transfer, the other direction is ignored (the bits are usually all 0 or all 1, depending on the idle state of the line).

Therefore, as long as everything is nicely in sync, you know that after sending a single word, you can also read exactly a single word.

But I guess it's not a good idea to rely on that alone. If you need to be sure that no data is left in the send buffer when you set CS high, you should check the busy flag. With embedded-hal 1.0.0, there will be a [flush](https://docs.rs/embedded-hal/1.0.0-alpha.9/embedded_hal/spi/trait.SpiBusFlush.html#tymethod.flush) method to do so. Embedded-hal 0.2 doesn't have such a method, so you need to wait for `is_busy()` to become `false`.

knutaf, 2022-12-12:
I'm not an expert here, but intuitively, putting data in the data register for sending seems like it needs to be an asynchronous operation, because the high and low values on the data line have to be synchronized with the SPI clock. What you're saying about the send buffer makes sense.

I guess my main remaining questions are:
1. would it be beneficial to have the behavior of this blocking write call be semantically the same as `spi_write_blocking`? i.e. automatically wait for not busy
2. or maybe it would be sufficient to document these nuances somewhere, perhaps on the blocking `write` function?

jannic, 2022-12-13:
Both are very valid questions!

As the `write` method is defined on the trait in embedded_hal, not in rp2040-hal, there's not much we can do. I don't see a way to add docs to that method from our crate.
And the embedded_hal already noticed that the docs on the old version are insufficient, and improved them for version 1.0.

Regarding blocking behaviour: There are two possible points of view. You are saying the blocking function should block because it makes sense from the caller's point of view in a common use case. But one could also argue that even a blocking function should block as little as possible. It is _allowed_ to block if it needs to (for example if the write buffer is full), but it's not _required_ to block, and it should only block if it's really necessary. That way, the caller can continue to do some useful work (eg. prepare the next bytes to be sent) while the buffer is draining. The downside of the latter is that, as you noticed, the caller needs to do additional checks before de-asserting CS.

I think those points will nicely covered by embedded-hal 1.0.0: The [SpiDevice](https://docs.rs/embedded-hal/1.0.0-alpha.9/embedded_hal/spi/index.html#device) abstraction handles CS, and properly waits for buffers to be flushed. And for the lower-level SpiBus abstraction, the documentation explains that manual flushing is necessary:  https://docs.rs/embedded-hal/1.0.0-alpha.9/embedded_hal/spi/index.html#flushing

knutaf, 2022-12-13:
I looked at the docs you linked and think that does a good job of addressing my concerns. There's no requirement to keep this API the same as the pico SDK, and as you point out, I've managed to file this issue on the wrong crate entirely, oops. :)

Thanks for the enlightening discussion, anyway.